mavlink/qgroundcontrol #3827: Survey flight suggestions;

AndKe, 2016-07-20:
1: Please label Start and end - and allow to "reverse direction" . (not only rotate whole mission using grid angle)   - While grid angle may rotate mission, it does not say anything about how/which direction the turns are mad in.  in the screenshot below, on the southern part, with a wind direction to south,  you will want every turn to be  towards wind, so the mission should go from right to left.

- if a mission in certain wind conditions may be close to an UAV's endurance, you will want to start far away, and be near a place to land and end.
-if a mission is partially in controlled airspace, you will want to start with the CTR part when cleared by ATC, (often in time of less traffic)  - then leave the area for the rest of the mission (rather than estimate when you will enter CTR, or hold for entering CTR)

2: There are now three types of altitude . Absolute, Relative, and (Relative to)Terrain .

3: drawing the secondary waypoints with bigger radius (as discussed in another issue) - gives the one that draws the plan a good idea of scale, and makes it easier to estimate actual turn radius and clearance to obstacles.

4: please add few arrows to indicate flight direction. (or waypoint numbers)

[image]

Trass3r, 2017-11-08:
Drawing the waypoints and flight direction arrows would be very helpful indeed.

DonLakeFlyer, 2017-11-08:
This issue is pretty old. Most of it is already supported. Except for:
* Terrain
* Direction arrows

Antiheavy, 2018-02-04:
> This issue is pretty old. Most of it is already supported. Except for:
> 
>     Terrain
>     Direction arrows

+1 for both of these

DonLakeFlyer, 2018-06-20:
* Terrain is now supported
* Direction arrows are covered in another Issue

closed 2018-06-20 by DonLakeFlyer