mavlink/qgroundcontrol #3837: QT: QML module not found, compiles just fine.

senosahisnu, 2016-07-21:
I'm having a problem with Qt5.5.1, as import QGroundControl module from qgroundcontrol.qrc not being found. Weird thing is, it compiles just fine. Since the modules are not found, therefore I cannot tinker around with QML Design. Is there any solution for this problem? Thank you for your attention.

DonLakeFlyer, 2016-07-22:
Qml Designer will never work with QGC qml files. They are way too complex for it to handle, also they need C++ controller code behind them for the to display correctly.

closed 2016-07-22 by DonLakeFlyer

wmx-github, 2016-07-30:
@DonLakeFlyer  I create my qml files and qmldir then import to  qrc. I  want to use my  qml  modules then  import to other qml files , tips "qml module not found" and xxx not installed (xxx means my qml module). Is there any solution ? Thanks

DonLakeFlyer, 2016-07-31:
If your changes fail at runtime then there is something wrong with either the way you have your resources set up or you qmldir. Can't determine what because I can't see your code. If the changes spit out an error at build time, but the built still succeeds you can usually just ignore it.

wmx-github, 2016-08-01:
@DonLakeFlyer  Oh, I konw. the Qt Creater IDE can find the qml file but the qml compiler can't. thus, I import the full path it works well.