mavlink/qgroundcontrol #3843: Mission upload keep current if currently in a mission

dagar, 2016-07-23:
When you upload a mission the vehicle is reset to the first waypoint. If you're running a mission and your current waypoint isn't deleted I think it would make sense to set it as current in the newly uploaded mission.

This can be done by setting current in MISSION_ITEM.
http://mavlink.org/messages/common#MISSION_ITEM

AndreasAntener, 2016-08-08:
@dagar @DonLakeFlyer I was just investigating this on the firmware side and the problem is that QGC always sets `current` on the first mission item. Stop gap measure would be to just don't set `current` by default (not sure what the logic there should be later)

DonLakeFlyer, 2016-08-09:
Hmm, interesting idea. Then just let firmware do whatever it thinks is the right thing. Will have to talk to ArduPilot folks about that.

dagar, 2016-08-09:
I think it'll be easier to capture user intention on the QGC side.
On the PX4 side we don't really have a clue if it's a slightly tweaked mission or completely new.
For example flying a mission and you placed a waypoint too far so you bring it closer and upload.

AndreasAntener, 2016-08-09:
@dagar in your example, if the user re-positions the waypoint it's currently targeting and uploads the mission again, it would then just use the new waypoint as target. Isn't that what we want? As I default behaviour I mean. But I understand your argument about a completely new mission. Just not sure if that's a frequent use case.
If we don't let the FCS keep the current target by default, we would need to make sure that QGC always knows the current waypoint the FCS it's targeting and visualizes that to the user when modifying the mission. Which IMHO seems more complicated and error prone (e.g. when the waypoint update from FCS didn't get through).

DonLakeFlyer, 2016-08-09:
The point more is to not set current if you want to just update the existing mission and let it keep going. My concern with trying to set current and update the mission is that it will be wrong by the time it gets to the vehicle for example. There kind of needs to be a "just update the mission" but not the current waypoint. 

QGC could also still support other options that would specifically set current.

AndreasAntener, 2016-08-09:
and that :)

dagar, 2016-08-09:
That was an example of one of the few times I think it's appropriate to keep current. Any significant change to me seems like starting from the beginning may be safer.

AndreasAntener, 2016-08-09:
@dagar oh ok

@DonLakeFlyer 
> There kind of needs to be a "just update the mission" but not the current waypoint. 

Makes me think of more than one upload button: "Upload and continue", "Upload and restart" (something in this direction). Could be something.

dagar, 2016-08-09:
I would certainly like those options, but most of the "normal" QGC users I talk to already think it's way too complicated.

DonLakeFlyer, 2016-08-09:
> I would certainly like those options, but most of the "normal" QGC users I talk to already think it's way too complicated.

Which part Daniel? Mission Planning actually has an "auto-sync" feature which I hid for 3.0 release. When you turn this on changes are synched immediately to the vehicle as you make then. That gets a bit more complicated when the vehicle is actually flying but I was planning on un-hiding that and working through the bugs. Does that help any?

DonLakeFlyer, 2016-08-09:
Or is it just the "planning" part that is bad. Adding items and so forth?

dagar, 2016-08-09:
More of a general sentiment from a few users. Sorry, I shouldn't make statements like that without more information. I'll try to gather useful data as these things come up.

In the auto-sync case is current preserved? Does anything cause it to reset to 0?

DonLakeFlyer, 2016-08-10:
Auto-sync without changes would work just like the user clicking Sync. No problem with making statements about confusion. Happy to figure out how to make it better. The more complaints the better to understand how to fix.

DonLakeFlyer, 2017-05-08:
This is all different now

closed 2017-05-08 by DonLakeFlyer