mavlink/qgroundcontrol #3849: Attitude rotation for tailsitter.

ivand58, 2016-07-25:
When the tailsitter/VTOL switches from MC to FW mode, the attitude should be rotated at 90 degree.

AndreasAntener, 2016-07-25:
@DonLakeFlyer about your comment whether this is a firmware or QGC issue: it depends.. Strictly looking from the firmware side, the attitude representation doesn't change during the tailsitter transition, it's just 90 degrees rotated. We're having the same problem in the plotting tools where we cannot use the logged euler angles anymore. On the other hand the firmware knows best what actually happens and how the user representation of the angles should be rotated. @tumbili and I are not sure yet what the best approach is.

DonLakeFlyer, 2016-07-28:
QGC really isn't involved in this sort of thing. To make it start to massage basic values like attitude which come back from the vehicle based on vehicle type seems like a really bad idea. That is way too much firmware specific/vehicle specific knowledge to bolt into a ground station. Not to mention that it would only work with a ground station that has very special code to handle it.

LorenzMeier, 2016-11-08:
@DonLakeFlyer That's actually incorrect. The vehicle attitude has nothing to do with the horizon indicator, which is something the GCS is responsible for.

We can help QGC by providing an user-facing attitude message.

closed 2026-10-04 by github-actions[bot]