mavlink/qgroundcontrol #3851: The unit in safety item doesn't change while changing the units

glody, 2016-07-25:
Test Environment
QGroundControl: Daily Build: HEAD:e6e0429 2016-07-20 18:58:20 -0400
Windows: Windows 8.1 64 Bit Laptop
Linux: Ubuntu 14.04 64 Bit Laptop
OS X: OS X 10.11.2 MacBook Pro
Purpose:
Check if the units in safety item is correct
Step:
1. Lanuch QGC
2. Change the units and relaunch QGC
3. Check the unit in safety item, setup page.
Excepted result:
The unit should change accordingly
Actual result:
The unit is still meter and centimeter
[image]
[image]
[image]

DonLakeFlyer, 2016-07-25:
Need to look a this for 3.0

DonLakeFlyer, 2016-07-25:
I thought this was plumbed through to parameters already but it isn't in 3.0. Just mission items get translation.

DonLakeFlyer, 2016-08-19:
Fixed now

closed 2016-08-19 by DonLakeFlyer