mavlink/qgroundcontrol #3853: Polygon adjustment bug

birchera, 2016-07-25:
When I move the map during adjustment of the polygon for surveying, the computed path is wrong. Waypoints seem to be computed way out of the polygon.

Before moving the map:
[image]

After moving the map:
[image]

DonLakeFlyer, 2016-07-25:
I'll try to repro this. But if you can save a mission file for both of these images it would save me time trying to repro. The first image also shows the grid generation where you have a non-parallel line in there. I've been having a hard time reproing that as well.

DonLakeFlyer, 2016-07-26:
This may be related to #3856.

birchera, 2016-07-26:
[initialMission.txt](https://github.com/mavlink/qgroundcontrol/files/383194/initialMission.txt)
[brokenMission.txt](https://github.com/mavlink/qgroundcontrol/files/383193/brokenMission.txt)

I found that the problem is the list of polygonPoints that goes into ComplexMissionItem::_gridGenerator(). Some of the points are Nan, causing the grid lines to be wrong.

Looking into why this is the case.

DonLakeFlyer, 2016-07-26:
Not related to #3856. For this bug the mission files don't help unfortunately since brokenMission.txt is trashed as well with a bad polyon area definition. The issue is a bad update while moving the map. But moving the map should not affect affect polygon coordinate values, only physical x,y position information. Hmm.

DonLakeFlyer, 2016-07-26:
It happens if a drag handle goes off the edge of the map. Could be a bad problem which means you can't really implement your own map positioning items in Qml.

DonLakeFlyer, 2016-07-26:
Looks like a Qt bug. Just installed Qt 5.7 and doesn't look to be fixed there unfortunately.

DonLakeFlyer, 2016-07-27:
I was wrong. This works fine in Qt 5.7. The issue is that Map.toCoordinate/fromCoordinate in 5.5 alwas clips to the viewport. Hence once the drag handles go outside the viewport they blow up. In 5.7 that is optional and should work correctly.

DonLakeFlyer, 2017-03-04:
Working correctly with switch to Qt 5.8

closed 2017-03-04 by DonLakeFlyer