mavlink/qgroundcontrol #3860: Launching Qgroundcontrol in Ubuntu

anthonydnguyen, 2016-07-26:
Hello guys,
I am new to robotics and linux. I am trying to learn.

I tried to launch Qgroundcontrol with ~/qgroundcontrol/deploy/qgroundcontrol-start.sh

[image]

Can someone help me?

DonLakeFlyer, 2016-07-26:
You are in the wrong directory. You need to be in your build output directory. Wherever the built executable ended up.

closed 2016-07-26 by DonLakeFlyer

anthonydnguyen, 2016-07-26:
I follow the instruction from http://qgroundcontrol.io/installing-qgroundcontrol/. After decomppressing Qgroundcontrol, they didn't mention about building it. How can i build the package?

DonLakeFlyer, 2016-07-26:
Ah, sorry. Thought you were building it yourself. You don't clone the repository you download the tar.bz2 and expand.