mavlink/qgroundcontrol #3864: Survey: Survey area vehicle entry

DonLakeFlyer, 2016-07-26:


birchera, 2016-07-26:
Some thoughts:

Candidate criteria could be
* Minimal mission distance
* Wind direction
* No-fly zones next to polygon (matters for vehicles with nonholonomic constraints)

Only the user has certain pieces of information, e.g. wind direction. This suggests that either we create inputs for all these parameters or give the user full control.

DonLakeFlyer, 2016-07-26:
For full controls I can see two options:
1) Flip entry/exit points into grid (in the example the vehicle would start the grid from the lower right "S" point)
2) Reverse the direction of the flight lines (in the example this would lead to the flight path of the first parallel line being from North to South instead of South to North)

Anything else?

DonLakeFlyer, 2017-12-29:
This can be controlled manually now. So good enough for now.

closed 2017-12-29 by DonLakeFlyer