mavlink/qgroundcontrol #3886: Update Debian / Ubuntu package and install instructions

LorenzMeier, 2016-07-30:
@DonLakeFlyer @dagar It would be good if we could update this PPA instead of the tar gz file. Managing the dependencies manually is a pain for the users.

https://launchpad.net/~qgroundcontrol/+archive/ubuntu/ppa

dagar, 2016-07-30:
I agree, we just need to find a way to automate it and determine how to handle Qt. In the past the Qt PPA we had no control over proved to be flaky. With a secondary flat version of the repo (no submodules) we could have builds handled by launchpad that would create PPAs for the full range of Ubuntu versions.
https://help.launchpad.net/Packaging/SourceBuilds/Recipes

For right now the best option on Linux is the Appimage. Just download and mark it executable.

    wget https://s3-us-west-2.amazonaws.com/qgroundcontrol/v3.0.0/QGroundControl.AppImage
    chmod +x QGroundControl.AppImage
    ./QGroundControl.AppImage

DonLakeFlyer, 2016-07-31:
@dagar Do you think you could update user guide to include the AppImage instructions for Linux?

dagar, 2016-07-31:
https://github.com/mavlink/qgc-user-guide/pull/25

tribbloid, 2016-08-27:
+1 ppa and auto-update will be nice.

probonopd, 2016-08-27:
Also see [AppImageUpdate](https://github.com/probonopd/AppImageKit/blob/master/AppImageUpdate.AppDir/README.md).

DonLakeFlyer, 2017-05-18:
@dagar Is this even valid now We aren't doing a PPA are we. Just AppImage and tar?

dagar, 2017-05-19:
It would be nice to have, but the launchpad stuff is surprisingly painful and still can't handle git submodules.  It also doesn't work very well with requiring a specific Qt version.

I think an Ubuntu Snap (https://build.snapcraft.io/, like an AppImage) would be better for QGC, and has a corresponding "store". 

I'll close this unless someone wants to dive into launchpad and bazaar.

closed 2017-05-19 by dagar

hamishwillee, 2017-05-22:
For what it's worth, 
1. I just tested and updated our installalation instructions.
2. I found the appimage instructions [easy to follow](https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html#appimage). I'm not a Linux expert so I think that these are sufficiently accessible for almost anyone.

Jaeyoung-Lim, 2018-05-17:
@dagar Not a follow up on this issue, but I saw there was an empty snapcraft.yaml file lying around and tried packaging QGC into a snap. [snapcraft.yaml](https://github.com/Jaeyoung-Lim/qgroundcontrol/commit/0e3369a3dc17de252a16d760287f1be7ae1cf085)

However, the current qmake plugin supported by [snapcraft only supports Qt 5.5.1](https://github.com/snapcore/snapcraft/blob/master/snapcraft/plugins/qmake.py#L92-L99)

Jaeyoung-Lim, 2018-05-17:
@dagar I did manage to package a working qgc into a snap using the prebuilt release (v1.3.3)

Would this be sufficient for now?

DonLakeFlyer, 2018-05-17:
Do we really need yet another way to package QGC on Linux? It's a hassle maintaining what we already have.

dagar, 2018-05-17:
@Jaeyoung-Lim sounds good to me.

@DonLakeFlyer Snap and AppImage area very similar things. The important difference from my perspective is that Ubuntu has a Snap "store" where we could now have QGC installable with a couple clicks and auto updated to the latest.

Jaeyoung-Lim, 2018-05-17:
@dagar Awesome. I am looking into some dependency fixes that needs to be done.

Once done, I will make a pull request with the snapcraft.yaml placed in the same location as the current master.

DonLakeFlyer, 2018-05-17:
Almost the same is not the same as exactly the same. The problem is these things get added and then folks disappear and then I'm stuck maintaining a crazy large set of things. All of which need to be tweaked in different ways when things like Qt versions change and so forth. It already takes hours or work to switch Qt versions or deal with the addition of a new Qt module. There has to be really high ROI to add yet more ways to release QGC.

dagar, 2018-05-23:
@DonLakeFlyer what I meant by that is I think we might be able to switch fully to Snaps.

DonLakeFlyer, 2018-05-23:
Will that work on non-Ubuntu Linux version?

dagar, 2018-05-23:
Yes - https://snapcraft.io/

DonLakeFlyer, 2018-05-23:
Ah, that would be ok then

TobiMiller, 2018-06-13:
You should definitely switch to snap because of the handling for the user. the update process would be so much easier.