mavlink/qgroundcontrol #3887: External compass orientation not available from start

julianoes, 2016-07-30:
I just wanted to calibrate my racer where the external GPS/mag is rotated but the autopilot is in the default position.

In the compass calibration view I was asked what the autopilot's orientation is but not what the external compass' orientation is. I had to do the calibration in order to get the second menu where I was allowed to select the orientation of the external compass.

In my opinion this second menu should be presented from the beginning, otherwise chances are high that I'll forget to set it after the calibration. Also, before doing the calibration I was wondering where I could set it.

[image]

DonLakeFlyer, 2016-07-31:
This is not possible. PX4 firmware only exposes the compass information:  available/internal/external after the calibration is complete. Prior to calibration QGC does not no anything about compasses. This will require firmware changes.

If you run ArduPilot firmware you'll see that it works exactly as you describe since that firmware detects compass info at boot time even if not calibrated.

julianoes, 2016-08-01:
Aha, so we could/should detect this on startup as well.

Should the detection be based on params or the MAV_SYS_STATUS_SENSOR message?

DonLakeFlyer, 2016-08-01:
QGC knows about compasses being there based on the CAL_MAG#_ID's having an id set on them. Firmware would need to set these up at boot time.

julianoes, 2016-08-01:
Aha, I'd guess that should be possible.

julianoes, 2018-02-22:
Stale.

closed 2018-02-22 by julianoes