mavlink/qgroundcontrol #3889: Android qgroundcontrol-beta from Play Store lost sik radio connection after a few seconds

wolkstein, 2016-07-30:
hi,
i just installed the bata version from google play store.
my device is an odroid c1 running android 5.1.1.
i use an sik 3DR Radio v2 from 3d robotics.
my copter use an pixhawk with latest stable ardupilot 3.4 firmware.

my odroid c1 can execute the tower-gsc and 3dr services connected via my 3dr sik radio without any problems.

but if i run the android qgroundcontrol-beta version from Google Play Store, the connection will lost after a few seconds. I can hit the reconnect button but the same thing happens. after a few seconds connection will lost.
if i am running the latest build of qgroundcontrol on my linux machine  the connection with the same 3dr radio works without any problems.

/g
wolke

DonLakeFlyer, 2016-09-04:
@wolkstein Does this reproduce with the Stable build that is now up on Google Play?

wolkstein, 2016-09-05:
i can not try, because i only find "qgroundcontrol (beta)"  on google play. is there an direct link to get the right apk.

dagar, 2016-09-05:
@wolkstein it's the same google play store listing either way. Use this link to control if you get the beta or stable version. https://play.google.com/apps/testing/org.mavlink.qgroundcontrol

birchera, 2016-09-16:
We observed the same problem. Specifically, the connection is already lost in a seemingly random moment during initial sync. In rather infrequent occasions this lead to a crash of the app. No problems were observed with the same setup on mac or ubuntu computers.

We have seen it on the same telemetry as @wolkstein, but also on the RFD868+.
Our tablet is a Samsung Galaxy Tab A running Android 5.0.2. We can already exclude a power issue, external powering of the telemetry has been tested.

On the plane side we have a pixracer, running px4 code.

We have tried both beta and stable versions, without success. To further investigate we were wondering, what the available tools are. We have used the extended logging, but this did not help. Unfortunately the single USB connector does not allow to plug in the telemetry and the computer. Is there a way to debug it using breakpoints or getting crash reports?

DonLakeFlyer, 2016-09-16:
So this only happens during initial connection? Is it during parameter load? The parameter load could be choking out th heatbeats. After it loses connection does it recover automatically?

birchera, 2016-09-17:
No, there is no automatic recovery. And yes, the logs show that it happens during parameter load. How could I check the hypothesis about the heartbeat?

Wouldn't it happen as well on a laptop?

DonLakeFlyer, 2016-09-17:
Let me try some testing. Initial comment talks about using Stable Copter 3.4 which doesn't really exist yet. Is this on Copter Beta 3.4 rc?

wolkstein, 2016-09-17:
sorry!  

> Initial comment talks about using Stable Copter 3.4
my fault. i use arducopter 3.3.3. i do not know why i wrote 3.4.

/g
wolke

birchera, 2016-09-19:
We tried to increase the time for the heartbeat time out [here](https://github.com/mavlink/qgroundcontrol/blob/9624d5c51c8407eeb4c36b80410e42a4d420dc9f/src/comm/LinkManager.h#L238). This has not yielded any success. If you could point me in the right direction @DonLakeFlyer, I can test some more. Thanks!

DonLakeFlyer, 2016-09-19:
When you get a communication lost, internally to QGC nothing really happens other than a UI display which allows you to disconnect. QGC still keeps running normally hoping it will get more messages from the vehicle (as long as you don't click Disconnect). If it does end up getting another heartbeat, the ui notification will just go away. So if the communication lost doesn't go away automatically after a long enough amount of time, it means that it has completely stopped getting heartbeats from the vehicle. That happens when either the firmware crashes or goes bonkers or there is something goes wrong with the comm link.

When you get the communication lost does parameter load complete? If parameter load completes, does QGC show HUD attitude changes while it says communication lost?

birchera, 2016-09-20:
Thanks for the explanation. It seems to actually lose communication altogether. The parameters don't finish the sync and there is no attitude information. Since also the firmware is fine, does this mean the problem is rather related to the telemetry/USB driver?

DonLakeFlyer, 2016-09-20:
Anyway to get debug output from the vehicle through the debug telem port? Wondering if the vehicle is reporting that it loses the connection.

birchera, 2016-09-21:
I did another test, this time not with a telemetry but with a serial link over cable. Without success.

Interestingly the firmware side does not report loss of connection, which changes once I unplug the cable. Then the commander module (px4) complains it lost connection.

I tested some of the logging options which looked related. The connection is well documented in the logs, as well as the transmission of various parameters. But nothing that would tell us anything about what is happening.

I am going to analyze the UART signal more closely tomorrow, but my fear is that something in / around the USB driver of the tablet messes up.

DonLakeFlyer, 2016-09-21:
@dogmaphobic Seems to point more and more to QGC/Android side getting screwed up.

birchera, 2016-09-22:
After some debugging it looks like lowering the baud rate did the trick. We were previously on 57600 and now on 38400. Will fly it in the next days to confirm reliability.

Thanks for the support.

DonLakeFlyer, 2016-09-22:
@dogmaphobic Gus, make any sense? Seems strange.

dogmaphobic, 2016-09-22:
No, it doesn't make sense unless the SiK radio is *autobauding* somehow. Isn't it fixed to 56k? If it is indeed setting the baud rate automatically and it works at a lower baud rate, it would indicate the probability of too much noise in its frequency band. But that would not explain Tower having no issues with it. In summary, I would say no, I don't get it...

I have not had the time to set aside to test this. I don't have anything using those (SiK) modems anymore. I will have to plug one in, set it all up and find an OTG cable to plug it to the tablet.

birchera, 2016-09-22:
First off, baud rate was configured to be 57600 on our modules.
Since you don't think this makes sense, let me fill you in with the rest of the things I tested today.
Initially I used a signal analyzer to look at the signals being sent through the FTDI wire. Interestingly signals kept being sent after connection was lost and nothing was reaching QGC anymore.

We then decided to try reducing the rate. After setting it to 38400 the problem was gone in 10 out of 10 attempts to reproduce it.

DonLakeFlyer, 2016-09-24:
I wonder if QGC is getting bytes but they are corrupted so it could enver decode into mavlink. @birchera Let me create a custom build for you to install that has some logging around data coming through the port.

birchera, 2016-10-02:
We tried the lower baud rate in flight, unfortunately communication was lost again during flight. Interestingly, this time it was always regained a bit later. This was not the case when we had the problem of initially transferring of parameters.
@DonLakeFlyer such a build with more logging could be very helpful. On which layer would you expect the issue to be? Is this in the SerialLink class or in the actual device driver?

sverling, 2016-10-03:
following up on the findings of @birchera I printed the raw serial data (in SerialLink.cc in the _readBytes function) with a timestamp. I also print the moment QGC sees the connection loss. There i observed the following:
At one point it does not get any serial data anymore. after something like 3-4 seconds QGC reports that the connection is lost. Also after that it doesn't receive any serial data. 
Only once i disconnected and connected again to the serial device it receives data again.

DonLakeFlyer, 2016-10-03:
Do you have the ability to debug/build android builds? Wondering it you can do the same thing (instrument) to see if bytes are comng through the android java serial port code. You would need to WiFi debug to the tablet and look at logcat output.

sverling, 2016-10-04:
I don't really have experience with java or android builds. This is why today we tried a setup where the ground station telemetry was connected to a bluetooth board. QGC was then connected through bluetooth. With that we did not experience the problem. That might speak for a serial driver issue or something similar.

DonLakeFlyer, 2016-10-04:
Ok, thanks. @dogmaphobic Is going to try to get some time to look at this.

DonLakeFlyer, 2016-10-11:
Same as #4055 I think

DonLakeFlyer, 2016-10-11:
@dogmaphobic Curious as to whether you looked at this instead: https://github.com/HelloZB/QtAndroidFTDI/tree/master/android. This seems to use the FTDI drivers from the FTDI folks instead of writing one by itself. Seems way simpler. I wonder it we are out of date with something. The code we have included into QGC seems pretty old.

DonLakeFlyer, 2016-10-12:
Can you try installing this apk: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/OneOffBuilds/QtApp-debug.apk. It may be a little flaky on disconnect, but it should conect.:

* Boot QGC
* Pluing Radio
* Connect to QGC "just once"

Hoping it will fix the flaky connection.

DonLakeFlyer, 2016-10-18:
@birchera Any time to try this build?

birchera, 2016-10-19:
Sorry, must have missed your post. Tested connection 5 times with resetting the cached params in between. Works like a charm! What did you change?

DonLakeFlyer, 2016-10-19:
Phew! That's good news. I rewrote the code for FTDI  connections to use the driver provided by the FTDI folks.

wolkstein, 2016-10-29:
hi,
i also test the app from https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/OneOffBuilds/QtApp-debug.apk
here the connection works very well. no disconnect or other problems. great!!

my environment is the same than some weeks ago if i open this issue.

- odroid c1+ running android 5.1.1.
- 3DR Radio v2 from 3d robotics.
- pixhawk with ardupilot 3.3.3 firmware

/g
wolke

birchera, 2016-10-29:
FYI @wolkstein: This fix has already been merged into master in https://github.com/mavlink/qgroundcontrol/pull/4157.

closed 2016-11-08 by DonLakeFlyer