mavlink/qgroundcontrol #3903: Survey area calculation is incorrect.

AndKe, 2016-08-02:
please see attached screenshot.
An area of about 1x1km is covered. (measured by Takeoff and 2 - being ~1km   , and the same distance of 1km between 2-3.

Calculated area is close to 100km2

Please correct calculation, and please consider presenting it in km2  , maybe with a decimal.
 (nobody really cares to see square decimeters anyway  even a small mission like 1302.52 m2   looks better as 1.3km2

[image]

birchera, 2016-08-02:
You can chose your prefered unit in general settings, where you have the choice of 5 different units.

I agree that for large missions the decimal digits of the square meters don't matter, but there may be other applications where the covered area is only very small.

Note that 1000m x 1000m results in 1'000'000mˆ2, which corresponds to 1kmˆ2. This is pretty accurately what the statistics show.

AndKe, 2016-08-02:
ok, this got very embarrassing very quickly, please consider making the km2 default - otherwise, I'll just close this issue and pretend like I never made such a mistake. :)

closed 2016-08-02 by AndKe