mavlink/qgroundcontrol #3913: Parameter editor: compare current parameter set against a params file

hsteinhaus, 2016-08-03:
APM Planner and Mission Planner (?) offer a feature that allows to compare the vehicle's params against a params file. This feature is very handy to quickly find out why a customer's vehicle is behaving "strange". Instead of manually reviewing 500+ params, this feature allows to concentrate on the dozen that is different from the initial values (which are not necessarily the default values hardcoded into the source).

IeiuniumLux, 2017-04-15:
Is this feature still scheduled to be in the next release or was this already implemented and it's included in the current daily builds?

DonLakeFlyer, 2017-04-15:
There is a dup (which I can't find now) of this somewhere which I moved to the next release. I'll keep this one as well. This won't be in the next stable.

closed 2026-10-04 by github-actions[bot]