mavlink/qgroundcontrol #3919: Crash after communication is lost on iOS

staroselskii, 2016-08-05:
Setup:
- iPhone 5S with iOS 9.3.2
- QGC Version 3.0.0 (2985)
- ArduCopter 3.4-rc1 on Navio 2 (```./arducopter -A udp:<ip>:<port>```)

How to reproduce:
1) Launch ArduCopter
2) Click "Connect" (via UDP)
3) Kill ArduCopter
4) Click on a popped-up "Communication Lost"- `Disconnect` button
5) Click on the QGC's icon in the top left corner

I'd be glad to provide you with some kind of a backtrace but I don't know how yet.

DonLakeFlyer, 2016-08-06:
Let me see if I can repro

DonLakeFlyer, 2016-08-12:
@staroselskii Couldn't repro in the iOS simulator using ArduCopter SITL. Curious as to why you need to manual connect to UDP as opposed to auto-connect? Maybe that has something to do with it.

staroselskii, 2016-08-13:
Auto-connect didn't work straight out of the box, so I've been using this
method instead. It'd be nice if someone with a real device could test it.

DonLakeFlyer, 2016-08-13:
What UDP settings are you using?

DonLakeFlyer, 2016-08-20:
Can you try the latest daily?

closed 2016-09-04 by DonLakeFlyer