mavlink/qgroundcontrol #3922: Android portrait mode

dagar, 2016-08-07:
We've already received a couple requests for portrait mode in Android. In portrait the fly screen is okay if we can do something with the attitude indicator/compass. Plan view is pretty much unusable. 
Something to think about going forward.

kellyschrock, 2018-12-03:
For what it's worth, I just did this, and QGC doesn't seem to have a problem running in portrait mode. Anyone have an idea of why it would have been restricted to landscape initially?

DonLakeFlyer, 2018-12-04:
> Plan view is pretty much unusable.

Nothing has changed with respect to Plan View being unusable. My guess is that there are other screens which won't work either. QGC is way more than just the Fly View.

kellyschrock, 2018-12-04:
@DonLakeFlyer FWIW, I didn't find the Plan view "unusuable". It wasn't as convenient as Landscape mode, but I was able to plan a mission with it and didn't run into anything I would call "trouble". I've been looking around at other screens and nothing has jumped out at me as being really messed up by portrait mode. The video feed looks truncated unless you specify an appropriate aspect ratio, but that's it. Thanks, by the way, for letting me know about the presence of screens other than the Fly screen. Super handy! Now I know that all those other screens I've been looking at in QGC aren't the Fly screen. :-)

DonLakeFlyer, 2018-12-04:
My guess would be Offline Maps in Settings is probably a problem as well. That's one that is screen size problematic. The issue with Portrait is the quality of the experience. Yes you can probably make things work if you are persistent and don't mind non-optimal experience. But the likely reaction on say Google Play with large set of users using Portrait mode is that the experience is crap.

DonLakeFlyer, 2018-12-04:
The interesting possibility would be a setting to "unlock" portrait mode which some sort of warning that you are entering the wild west of user experience. So default would be off.

kellyschrock, 2018-12-04:
That is an interesting possibility. Maybe put a note next to the option
indicating there could be a reduced UX if that option is turned on.

I was expecting maps to be a problem too, but they seemed to load pretty
well. Honestly, I was surprised at how well QGC reacted to suddenly being
in portrait mode in general. With Android apps, there's a fair bit of work
on every screen to allow that to work properly. Yet another reason to
consider QML for my next Android app. :-)

closed 2026-10-04 by github-actions[bot]