mavlink/qgroundcontrol #3923: The characters in some pages is unclear by default font size

glody, 2016-08-08:
Test Environment
QGroundControl: Daily Build: HEAD:2a610ae 2016-08-02 12:05:50 -0700
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the display in every pages are show normally
Step:
1. Lanuch QGC
2. Connect to IRIS+
3. Check the charachters in some setup pages
Excepted result:
The characters in every pages all show normally
Actual result:
The characters in some pages is unclear, change the font size bigger, the characters will clear.
[image]
[image]
[image]
[image]

DonLakeFlyer, 2016-08-09:
More windows font weirdness. @NaterGator Hadn't you figure this out when it was out of whack in Airframe?

DonLakeFlyer, 2016-08-18:
@NaterGator Didn't you fix something like this ages ago on Windows? Don't remember what the fix was.

DonLakeFlyer, 2016-12-24:
This is the windows positioning problem. Nate has most of this fixed. Full 100% fix scheduled for 3.2 on separate issue.

closed 2016-12-24 by DonLakeFlyer