mavlink/qgroundcontrol #3924: The information in 'Status Details' is incorrect

glody, 2016-08-08:
Test Environment
QGroundControl: Daily Build: HEAD:2a610ae 2016-08-02 12:05:50 -0700
Windows: Windows 8.1 64 Bit Laptop
Purpose:
Check if the status details shows correctly
Step:
1. Lanuch QGC
2. Connect to IRIS+
3. Check the 'Status Dateils' by click widgets drop-down menu
Excepted result:
The information in 'Status Details' should shows according the device data
Actual result:
All data are 0.00
[image]

DonLakeFlyer, 2016-08-09:
I'll take a look, but that whole widget is going to go away in 3.1. So may just toss it all instead of fix.

DonLakeFlyer, 2016-08-18:
I pulled this out

closed 2016-08-18 by DonLakeFlyer