mavlink/qgroundcontrol #3925: UAVCAN node parameters do not appear correctly

mhkabir, 2016-08-08:
The parameters from connected UAVCAN nodes should each appear under a different heading per node. 

Instead it's all mishmashed under Default Parameters and impossible to use. 

@pavel-kirienko will be able to better explain where the issue might lie.

pavel-kirienko, 2016-08-08:
Thanks Kabir. All I can tell so far is that parameters appear to be sent correctly from Pixhawk, each associated with the node ID it was retrieved from. I would hazard a guess that the node ID is probably being ignored in QGC.

bendyer, 2016-08-09:
Bug in the MAVLink/UAVCAN bridge; see https://github.com/PX4/Firmware/pull/5271#issuecomment-238399392

DonLakeFlyer, 2016-08-09:
The grouping of parameters is controller by firmware meta data. I don't have any UAVCAN hardware so I can't look at this myself. An image would be helpful to determine if there is really anything wrong on the QGC side.

closed 2026-10-04 by github-actions[bot]