mavlink/qgroundcontrol #3927: qgroundcontrol cannot run multiple instances and collides with other applications on port 14499

tfoote, 2016-08-08:
We have been working on a demonstration of a swarm of drones and we ran into a port collision. Surprisingly qgroundcontrol refused to start when port 14499 was open. 

I found the port 14499 is selected here: https://github.com/mavlink/qgroundcontrol/blob/a8fda9bc69a04453ec008e61c7d1d0d89f27b65f/src/main.cc#L29

And a little bit below it's used [here]( https://github.com/mavlink/qgroundcontrol/blob/a8fda9bc69a04453ec008e61c7d1d0d89f27b65f/src/main.cc#L113-L121)

This appears to have been requested here: https://github.com/mavlink/qgroundcontrol/issues/2524 and implemented here: https://github.com/mavlink/qgroundcontrol/pull/2703

And all it does is quit if that port is not available. 

We ran into this initially as we are using a large block of ports starting at 14000, and we can clearly move the block that we're using to avoid this port. 

We didn't run into this yet as we're usually running one QGC instance connected to both parts of the swarm, but this behavior will become a problem in the future as we expect to have segmented access to the parts of the swarm for teams. 

There's not a lot of background in #2524 as to why this behavior was added. Are there technical problems with running multiple instances of QGC or is this to avoid user mistakes?

Brainstorming possible solutions:

Can we change this to be a warning/prompt that the user can choose whether to quit, retry or continue anyway? What about a command line option to either change the port or disable the check? Switch to use QtSingleApplication ([source](https://qt.gitorious.org/qt-solutions/qt-solutions/trees/master/qtsingleapplication?p=qt-solutions:qt-solutions.git;a=tree;f=qtsingleapplication;h=6d02dafce404509aaa3725d5fdc5ec5b898465c5;hb=HEAD)) or [similar Qt based cross platform method](https://stackoverflow.com/questions/5006547/qt-best-practice-for-a-single-instance-app-protection). 

I'd lean toward the last option with a prompt when it's triggered to prompt the user to shutdown the other instance or override or quit and a command line option to force any of the above 3 behaviors.

DonLakeFlyer, 2016-08-09:
I"ll take a look at the other possibilities you posted to change this.

DonLakeFlyer, 2016-08-09:
Also QGC does not support multiple instances. If you want multi-vehicle, just connect multiple vehicles to the same QGC instance.

closed 2016-12-01 by DonLakeFlyer

paan2097, 2017-08-31:
I need to run multiple instances of QGroundControl or multiple HIL configs. I have two separate instances of XPlane up and running, but now need to use either two HIL configs or separate QGroundControls to run multple HITL simulations on a single computer. Can anyone advice on how I can get this working? Thanks.

piyushchauhan, 2019-12-14:
> Also QGC does not support multiple instances. If you want multi-vehicle, just connect multiple vehicles to the same QGC instance.

What can I do when I want to open separate window of each UAV and monitor them continuously?

aiegoo, 2022-03-22:
you can use virtual environment, such as docker or VMWare. or did you find any solution for launching multiple instances of QGC?