mavlink/qgroundcontrol #3934: Add FW landing complex mission item

dagar, 2016-08-12:
Opening this as a note to myself, but any feedback if others are interested would be great.

A nice fixed wing landing is
MAV_CMD_DO_LAND_START
MAV_CMD_NAV_LOITER_TO_ALT
MAV_CMD_NAV_LAND

It would be nice if this could be added as one complex mission item with a sensible default altitude for the loiter and then the distance between loiter and land based on FW_LND_ANG. Ideally you could position and rotate them all together.

DonLakeFlyer, 2016-08-12:
Great idea for new complex mission item. I still need to fix ComplexMissionItem so it's reusable. Hardwired to Survey now.

Antiheavy, 2017-01-31:
I am very interested in this functionality as the default landing method for my fixed wing project.

Additionally:
a) I think it is very important that fixed wing RTL is tied to a valid landing pattern such as this.
b) the mission validity checker should require a valid landing pattern such as this prior to accepting mission upload.  Or some other method of ensuring the aircraft will land as intended during lost comms and other RTL failsafes.

I am happy to discuss details and ideas on how this might be implemented from a user perspective.

Antiheavy, 2017-10-17:
hey, this excellent feature has been in QGC for a while now.  does this still need to be open?

DonLakeFlyer, 2017-10-17:
Nope

closed 2017-10-17 by DonLakeFlyer