mavlink/qgroundcontrol #3939: support for OpenLRSng

gregd72002, 2016-08-13:
Hi,

Is there currently any support for PPM output, or any support planned in the near future?

Are we able to plug an OpenLRSng transmitter through a Serial device and pass RC request through PPM as opposed to MavLink?

If not, one way of doing it would be through an auxiliary Arudino Micro which serves as a proxy between QGroundControl serial output and OpenLRSng device. Whenever an RC request arrives it would convert it into PPM while proxying all the other messages from QGC to OpenLRSng.

DonLakeFlyer, 2016-08-16:
Seems like a pretty special case usage pattern as opposed to the standard MANUAL_CONTROL mavlink mechanism.

gregd72002, 2016-08-16:
Maybe it is. But how do you currently do long range flights with Joystick plugged to QGC?

Isn't the standard long range telemetry radio too slow to send all MANUAL_CONTROL messages?

DonLakeFlyer, 2016-08-18:
Don't now. Not sure anyone is doing long range stuff via joystick. Seems like at some point it all gets so complicated to setup you might as well just use a regular radio.

closed 2016-10-04 by gregd72002