mavlink/qgroundcontrol #3940: warning popup doesn't update till you try to close it (development version)

kd0aij, 2016-08-13:
e.g. box appears showing "reject posctl" (with X in upper right hand corner): click the X and instead of closing, it displays the next warning message

DonLakeFlyer, 2016-08-13:
That's the way it's supposed to work. Otherwise you may miss the fact that a new error message got added while the old one was up.

NaterGator, 2016-08-13:
This has bugged me, there's got to be a better way to do this. What if the message that is waiting to display is more critical? 

Maybe we allow messages to aggregate in a single dialog until they take up a certain percentage of the screen? I don't really know what the right UX here is, but I think we need to think hard about this one.

kd0aij, 2016-08-13:
Yes, it never makes sense to hide warnings in a realtime system.

DonLakeFlyer, 2016-08-13:
> Maybe we allow messages to aggregate in a single dialog until they take up a certain percentage of the screen?

I specifically didn't do that since I thought they might not be noticed by just adding to an existing popup. Hence new messages that come by while a popup is already up go into the next popup. Additional messages are not hidden, they are just delayed which is likely just as bad.

Happy to change if anyone has a better idea.

kd0aij, 2016-08-13:
I like Nate's suggestion