mavlink/qgroundcontrol #3947: Cherry Pick to Stable V3.0.1

DonLakeFlyer, 2016-08-16:
https://github.com/mavlink/qgroundcontrol/pull/3936
https://github.com/mavlink/qgroundcontrol/pull/3930

DonLakeFlyer, 2016-08-29:
@dagar Are you good with cherry-pick? I certainly am not. Could use help to get these pulls over to Stable_V3.0 and cut a V3.0.1.

dagar, 2016-08-29:
Sure I can take care of these right now. All you need is a good diff tool.

DonLakeFlyer, 2016-08-29:
Ah, I think I could have figured it out that way! I was trying to use git cherry-pick and I was making more of a mess than anything else!

dagar, 2016-08-30:
I still cherry-picked each, but had to resolve a few bad conflicts.
https://github.com/mavlink/qgroundcontrol/compare/Stable_V3.0...Stable_V3.0.1?expand=1

One thing that would help if you think we'll be doing a lot of backporting in the future is merging PRs with the "Confirm squash and merge" button. This creates a single commit from each pull request with no merge commit.

DonLakeFlyer, 2016-08-30:
Compare looks fine. I was thinking this would just go into V3.0 branch and we tag to 3.0.1. Not a whole new branch. I think a whole new branch would only be for V3.1. Make sense?

dagar, 2016-08-30:
Makes sense. I'll push Stable_V3.0.1 to Stable_V3.0 and then delete it.

dagar, 2016-08-30:
I'll try the linux build before tagging v3.0.1. Is there any testing you'd like to do?

DonLakeFlyer, 2016-08-30:
Yeah let me test the other installers. Then we I give the ok sign, you can tag.

dagar, 2016-08-30:
Changelog?

 - **active binding fixed?**
 - Windows GStreamer dependencies now included properly
 - properly convert knots to m/s
 - **Move battery.percentRemainingAnnounce to Qml Global ????**

DonLakeFlyer, 2016-08-30:
* Attitude indicator correctly updates when displayed in horizontal orientation instrument panel (small phone screens, virtual joystick turned on)
* Video working correctly on Windows
* Properly convert knots to m/s
* Fix crash on Vehicle disconnect

closed 2016-09-14 by dagar