mavlink/qgroundcontrol #3948: Plan view show waypoint altitude

dagar, 2016-08-16:
What do you think about adding altitude to each collapsed waypoint in the fly view?

[image]

DonLakeFlyer, 2016-08-18:
FYI: I've been thinking of getting rid of the right hand list and just showing the currently selected item. If I do that this would go away. Still haven't decided.

dagar, 2016-08-18:
I think that could ultimately be better, especially if the appearance of the waypoints could change. Displaying loiters, different icons for commands, etc.

DonLakeFlyer, 2016-08-18:
> I think that could ultimately be better, especially if the appearance of the waypoints could change. Displaying loiters, different icons for commands, etc.

Yeah, exactly.

tubeme, 2016-08-19:
@DonLakeFlyer Just remove this waypoint list. It clutters the map. The way QGC is going is WYSIWYG so i really need to click on the point I want and the waypoint edit box should pop up. Nothing more. But for professional work I need another LIST view or plane where I need a table view of the mission! With all the attributes, height total, rel, coordinates of points etc. etc. And the option to edit the table. So If I want to change 20 points height should be matter of 30 seconds, not 5 minutes clicking way points.

Also @dagar very good idea about the height written on the point. During my mission planning sometimes I want to see the mission plot and be able to imagine where and how the plane will rise and fall and fly. Now I see the height diagram on the bottom but visually it is hard to interpolate it over the dots. 

Because except relative height I would love to see a total height in every waypoint. Also I would love to see the speed as well also the acceptance radius or other attributes. Like I can make camera automation and then want to see it how it is set during the mission in every point. So we can have a dynamic approach to visualizing the waypoints on the map. We push a soft button and toggle relative height. So we see beside every waypoint its relative height. We toggle total height, or other information. We can extend the visualization with time. Adding new ways for the pilot to help visualize its mission over the map. Analyze it "in need" fast (during mission execution), correct it during flight and re-execute it. We should be able with the flip of a couple of button to change map views in ways that help us work in dynamic situations. 

Also for the mission total calculations use Hours, Minutes format and Kilometers, Meters. Not just meters or minutes.

@dagar The idea for different Icons is very good indeed. Especially if we have complex mission with multiple of survey sites at one flight. You get lost in all the same waypoints.

NaterGator, 2016-08-19:
A separate tabular view of all waypoint data would mesh well with the stale PR I started [here](https://github.com/mavlink/qgroundcontrol/pull/3039). I am in agreement though, the flickable list of waypoint items is cluttered and difficult to navigate. Much better to use gefingerpoken.

DonLakeFlyer, 2016-11-25:
Rework for this moving to 3.2