mavlink/qgroundcontrol #3949: Crash after communication is lost on Android (and Ubuntu)

mswingtra, 2016-08-17:
Setup:

   -  Samsung Galaxy Tab A 9.7
   -  QGC Version 3.0.0-139-g801fdde
   - Pixracer v1, telemetry over serial port

How to reproduce:
1) Launch Pixracer
2) Connects automatically on Android
3) Kill Pixracer
4) Click on a popped-up "Communication Lost"- Disconnect button
5) Click on the QGC's icon in the top left corner

It is very similar to https://github.com/mavlink/qgroundcontrol/issues/3919. 
It also happens on Ubuntu 14.04, same steps required to reproduce it.

Short debugging we have done on our side on Ubuntu version showed that it fails in Fact.cc in line 154.

DonLakeFlyer, 2016-08-19:
Thanks for the great bug report. I can repro. This one has been hanging around for a while. Working on fix.

DonLakeFlyer, 2016-08-20:
Can you try the latest daily?

mswingtra, 2016-08-24:
Tested on Android and Ubuntu. I cannot reproduce it any more. Thanks a lot for the fix!

closed 2016-08-24 by mswingtra