mavlink/qgroundcontrol #3952: Log download and onboard files not working on release 1.4.1 (Pixracer)

kd0aij, 2016-08-18:
1. release 1.4.1 firmware: new logger files (extension .ulog) don't appear in the log download window

2. The onboard files widget reports:
[image]

DonLakeFlyer, 2016-08-18:
List files returns crazy repeated results for me before timing out.

@kd0aij Does Log Download widget work for you as a workaround?

kd0aij, 2016-08-18:
No, "new logger files (extension .ulog) don't appear in the log download window" so that function is not usable. Which method would be faster, if they both worked?

DonLakeFlyer, 2016-08-18:
Since Log Download was implemented the Onboard Files hasn't had much work on it. I think Nate has sped up Load Download so that it is about the same. Sounds like the firmware side implementation of Log Download needs to be updated due to the extension change. Somebody needs to fix that on the firmware side.

dogmaphobic, 2016-08-19:
Here is where the log file is identified:

https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_log_handler.cpp#L536

If nothing happens between now and tomorrow, I can take a look at the firmware side.

dogmaphobic, 2016-08-19:
Support for the new log file extension is here: https://github.com/PX4/Firmware/pull/5355

kd0aij, 2016-08-19:
ulog file download works with #5355

I'd never used this widget before, but my immediate observations are:
- The "Log Download:Download" function would be easier to use if it remembered your directory selection.
- The Erase function can blow the logger's current subdirectory out from under it. It's necessary to reboot the autopilot to recover (logger stop/start sequence results in a hardfault). (obviously also a firmware issue)

kd0aij, 2016-08-20:
Downloading .ulg files results in files named "log_N_UnknownDate.px4log
The extension **must** be ".ulg" or FlightPLot will fail to load the file. 
What is the reason for renaming the file anyway? Is it not possible to retain the directory structure from the SD card?

dogmaphobic, 2016-08-20:
No, the MavLink protocol doesn't have a file name. Just an index, date and size. That could obviously be recreated on the QGC side. The same for the `.px4log`. That's done on the QGC side and it would just a matter of fixing it there.

>Downloading .ulg files results in files named "log_N_UnknownDate.px4log

Would it be possible for you to show how these files are on your SD card? I don't have any functional "hardware" at the moment. I've been using SITL for these. Just mount the SD card and do a directory listing on one of the log directories so I can see how they look. Specifically, the ones that are showing up as `log_N_UnknownDate`.

Thanks!

dogmaphobic, 2016-08-20:
> The "Log Download:Download" function would be easier to use if it remembered your directory selection.

It should do that. If it doesn't it's a bug.

> The Erase function can blow the logger's current subdirectory out from under it. It's necessary to reboot the autopilot to recover (logger stop/start sequence results in a hardfault). (obviously also a firmware issue)

That's a bit more involved. I was under the assumption the log only runs once the vehicle is armed.

Either way, it would be best to create separate issues for these.

kd0aij, 2016-08-20:
If you don't have a GPS fix, this is what you get: (session number increments on boot, log number increments on arming)
```
nsh> ls -lR /fs/microsd/log
/fs/microsd/log:
 drw-rw-rw-       0 sess001/
 drw-rw-rw-       0 sess002/
/fs/microsd/log/sess001:
 -rw-rw-rw-  299033 log001.ulg
 -rw-rw-rw-  239986 log002.ulg
/fs/microsd/log/sess002:
 -rw-rw-rw-  235110 log001.ulg
 -rw-rw-rw-  164588 log002.ulg
```

kd0aij, 2016-08-23:
Another observation which might be relevant to log download: Bebop V1 video files transfer over USB3 at 19 MB/sec, which is 2 orders of magnitude faster than PX4 log downloads.

DonLakeFlyer, 2016-09-23:
@kd0aij What is the state of this Issue? Can't quite determine if there is still a problem. Also FYI: Onboard files only works over USB connection and even with that is still buggy. But that's covered in another issue already. Onboard files has kind of been replace by Log Download widget it hasn't gotten any fixes.

kd0aij, 2016-09-23:
@DonLakeFlyer @dogmaphobic It looks like no work has been done since my last 3 or 4 comments.

DonLakeFlyer, 2016-09-23:
Is it just that the file names are wrong when you download the new log file type? Or is there something else?

kd0aij, 2016-09-23:
As I said here: https://github.com/mavlink/qgroundcontrol/issues/3952#issuecomment-241208640 it would also be nice if the log directory structure were retained. I assume you'll find all your log files currently still have a px4log extension even if you download "new" logger files.

dogmaphobic, 2016-09-23:
As far as the Log Download widget is concerned, as I explained above, the flight controller only sends an index (log # n), a time stamp and a file size. There is no way for QGC to know what the file was named or its position within some hierarchical structure (other than infer it based on the date, which would lead to bunching up unrelated files that happen to have no time recorded). The extension used is arbitrary. Are we talking about the Log Download widget or the FTP tool?

kd0aij, 2016-09-23:
Log Download widget: https://github.com/mavlink/qgroundcontrol/issues/3952#issuecomment-241129120

OK, so there's no way to handle both .px4log and .ulg files or to replicate the directory structure, and fixing the extension will have to wait until sdlog2 is completely replaced by the new logger. At least it's a practical method for copying logs without removing the SD card, and manually renaming files is straightforward.

I guess there's an obvious reason these file transfers would be 100 times slower than with Bebop? Maybe a slow SD card driver in NuttX?

DonLakeFlyer, 2016-09-24:
Closing this.

I can say why download is so slow. But whatever it is QGC has no control over that. The packets stream to QGC in chunks without QGC needing to request each one separately. So there isn't much of anything QGC can do on it's end. The only delays imposed by QGC would be caused by timeouts diring due to missed packets. But over USB that doesn't ahppen that much.

closed 2016-09-24 by DonLakeFlyer