mavlink/qgroundcontrol #3956: PPM Radio Controller (as Joystick) via USB on QGC

CaptainJL, 2016-08-19:
Hello all,

I have been having a problem with interfacing the Turnigy (8 channel) radios to act as a Joystick in QGC and need help solving this issue.

Radio Equipment:
Both Turnigy 9X and 9XR radios.
PPM to USB Simulator Lead for connected Radios to PC (http://www.hobbyking.com/hobbyking/store/__24348__USB_Simulator_Lead_for_Turnigy_GTX3_Transmitter_VRC_Sim_Compatible.html)

Purpose:
I need to interface the Turnigy radio with QGC, so it acts as a virtual joystick.
I need to simulate a quadrotor using the PX4 dev code on Gazebo (on Ubuntu 14.04), which requires a joystick interface via QGC.

Problem Details:
The output from the PPM-USB converter is not large enough for QGC to be able to use (and thus will not calibrate). The min, centre and max positions on QGC are shown in the attached images 'low_n', 'mid_n', 'max_n'.

[image]
[image]
[image]

QGC also will not recognise Ubuntu's calibrations for joysticks.
I have pre-calibrated via both 'jscal' and 'jstest-gtk' on the Ubuntu System and it shows it's calibrated (which shows a range from 60-140 (100 centre)). This is shown in the attached image 'js_cal'.

[image]

I have spent a day on trying to solve this with no success and not sure what I should do.
Has anyone got any idea or has been able to solve this QGC PWM reading issue?

Thanks,
Jean-Luc Stevens

DonLakeFlyer, 2016-08-19:
Can you do the following:
- Turn on logging for JoystickConfigControllerLog (https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/Support/Support.html)
- Start joystcik cal with your Turnigy
- Just try to get past the first step of Throttle up
- After it fails to work, Cancel the calibration
- Save the console log and post it somewhere public I can look at it

CaptainJL, 2016-08-22:
Hey Don,
You can find the .px4log stored at: https://github.com/CaptainJL/QGC_Cal
Thanks,

DonLakeFlyer, 2016-08-22:
That is px4 binary log. I need a console log from QGC Settings/Console.

CaptainJL, 2016-08-23:
Oh sorry, I misinterpreted it what you needed.

I have uploaded the actual console log to: https://github.com/CaptainJL/QGC_Cal (as console_log).
For reference:
Axis 0: RHS horizontal (roll)
Axis 1: RHS vertical (pitch_
Axis 2: LHS vertical (thrust)
Axis 3: Unknown (Not a true axis or button)
Axis 4: LHS horizontal (yaw)

Thanks,

DonLakeFlyer, 2016-08-26:
This setup is sending very strange joystick values. Min Throttle is -3000 or so and full throttle up is 0. That's going to be too strange of returned values for QGC to handle or even to figure out some workaround.

CaptainJL, 2016-08-27:
Yep that is exactly what is happening. I do have a suggestion to fix it however (as I could calibrate correctly this way, just cant use it on QGC).
In a later GQC release, let it access the joystick calibrations settings that you can make on Ubuntu itself (eg. via jscal, etc).
Most Ubuntu software can already use it, such as flight sims, etc.

Is there a way I could add this to a suggestions list?

DonLakeFlyer, 2016-08-27:
Two problems:
1) It way too much of an OS specific change
2) Radio Cal does way more than just calibrate end points. It detects which stick is which function, reverse setting and so forth.

DonLakeFlyer, 2016-10-18:
There isn't going to be a generic way to fix this. Reading OS specific calibration settings can't be done since QGC needs to work on all OS.

closed 2016-10-18 by DonLakeFlyer