mavlink/qgroundcontrol #3961: Translation Support

dagar, 2016-08-20:
Through the Google Play listing I've been contacted by multiple Chinese users interested in adding translation support.

It looks like we wouldn't need much to get started, and then these users could help with pointing out missing or incorrect translations.

http://doc.qt.io/qt-5/internationalization.html

dagar, 2016-08-20:
Also relevant. https://github.com/mavlink/qgroundcontrol/issues/2882

dogmaphobic, 2016-08-20:
I used this service before (for a large Qt project) and really liked it. They support Qt directly:

https://crowdin.com

dogmaphobic, 2016-08-20:
Note that there is great deal more involved than just translating strings. Handling of Unicode on Windows for example is a nightmare. Dialogs, buttons, etc. will all have to handle the difference in size (and direction of writing as appropriate).

dagar, 2016-08-20:
Yes, I was afraid that would likely be the case. Is there an intelligent way to attack the problem or do we just enable and fix the resulting issues?

dagar, 2016-08-20:
I've submitted the crowdin open source project request.

dogmaphobic, 2016-08-20:
You start translating and deal with the issues as we bump into them. QGC is not as verbose as *regular* application. Dialogs should be simpler to accommodate based on the string lengths. The main issue I see off the top of my head are places that expect ASCII strings. These need to be converted. On Mac OS, that's automatic. On Windows, you need to go through some strange conversion for each call that expects an ASCII string, but once you find the pattern, it's just a matter of going through the code and replacing the calls with a call through the conversion function. I don't think we will be translating to Hebrew (or other right-to-left language) any time soon so that's something to worry about later.

dogmaphobic, 2016-08-20:
For example, things like this:
https://github.com/mavlink/qgroundcontrol/blob/master/src/FactSystem/FactGroup.h#L61-L70
Would require special handling in Windows (not sure about Linux). Mac OS you can go straight from QString to char but not in Windows.

dogmaphobic, 2016-08-20:
There are straight forward dialog text strings (UI). These are the simplest to tackle. Then you have things that come from parameter files (XML, Json, etc.) The user facing strings need to be translated and converted. And then you have things you have no (direct) control such as messages that come from the firmware. Those come in as straight (English) ASCII.

DonLakeFlyer, 2016-08-21:
Daniel: Could you look into how we would need to set up the build system for translations?

x923386210, 2016-08-22:
I am a Chinese user. I am happy to extend QGC to China. Translation support can be carried out.

x923386210, 2016-08-22:
I think the default application is consistent with the language of the mobile phone system. Also provide language switching option is relatively good. Android development I know how to do. Application and system language to maintain consistency does not know the method for Qt on Android.

dagar, 2016-08-24:
https://crowdin.com/project/qgroundcontrol

vooon, 2016-08-24:
@dagar i can help with Russian translation.

dagar, 2016-08-31:
Thanks @vooon 
I'll let you know when we have something to test/verify.

jilinggui, 2017-04-21:
I encountered a strange question, when I  QGC （Part of the translation into Chinese）installed in the android 6 tablet, the work is normal, but when I installed to the android 7 tablet, the show garbled

DonLakeFlyer, 2017-04-24:
Not quite sure what you are saying. Are you saying that you translated some of QGC into Chinese and it isn't working correctly on Android 7. Or that you are using a standard QGC build that you downloaded and it isn't working on Android 7.

GlidLov, 2018-05-10:
i can help with Italian translation

hamishwillee, 2018-08-21:
@dagar @DonLakeFlyer 
- What is the status of this? 
- Did you end up adding a translation project on crowdin? 
- Did you resolve the localisation issues for QGC itself (text direction etc.)

We're getting interest in translating again, so I'd like to have a document on status and "how to onboard" in the dev guide.

Just FYI, we are going to use Crowdin going forward for docs. It seems to have good support for the .ts strings out of Qt, and XML (see https://github.com/mavlink/qgroundcontrol/issues/3961#issuecomment-241209489 ). AFAIK no system does particularly well when working with plain text files, but crowdin at least makes a reasonable effort of identifying strings and allowing you to reuse translations.

Thoughts?

DonLakeFlyer, 2018-08-21:
@dogmaphobic Is the translation person. I don't know much about the status.

dogmaphobic, 2018-08-28:
Yes, it's all set and ready to go. We just haven't had much (translation) work done. It's been there for about 2 years. There has been some work done for French but not much else.

https://crowdin.com/project/qgroundcontrol

At this point, I would need to update the strings as it's been a couple of months since the last time I did it. But other than that, it's all set to go. We just need people to contribute with localization work.

hamishwillee, 2018-08-29:
@dogmaphobic FYI, I added info to the devguide on translation: https://github.com/mavlink/qgc-dev-guide/pull/44

Presumably this issue can now be closed?

dogmaphobic, 2018-09-03:
I've updated the current (master) strings: https://github.com/mavlink/qgroundcontrol/pull/6844

I've also updated crowdin with the new strings (443 strings added and 224 deleted)

hamishwillee, 2018-09-03:
Thanks Gus. @bys1123 - do you think you might be able to give us an estimate when the Chinese translation might be ready? ie 25, 50, 80, 100%?

dogmaphobic, 2018-09-07:
No, this is the first time we have something that can actually be added to QGC. Adding localization support will only take me a few minutes once I have the files. I just looked at crowdin and could not find the localization you mentioned above (TR). Is it there?

https://crowdin.com/project/qgroundcontrol

If you did the localization on your own computer, you can submit a PR with it but it will be hard to maintain as all localization work is meant to be done within crowdin. If there are changes to the strings, which happens on a daily basis, you will have to keep track of it yourself.

I've created a branch (and its PR) with the code to load localization strings. You can issue a PR against that branch (with your localization files) or wait for it to be merged to master.

https://github.com/mavlink/qgroundcontrol/pull/6859

hamishwillee, 2018-09-10:
@dogmaphobic https://github.com/mavlink/qgroundcontrol/tree/master/localization doesn't mention our "desired" translation system via crowdin. Ie we want to make sure that people use crowdin where possible because that system will maintain the strings and "know" when something changes, while this sort of thing is unmaintainable. 

Can you update that doc?

dogmaphobic, 2018-09-10:
Yes, that needs a change of focus. When I first put that together, it was meant for local maintenance of files coming from crowdin.

hamishwillee, 2018-09-10:
> Yes, that needs a change of focus. When I first put that together, it was meant for local maintenance of files coming from crowdin.

OK. We could also move it into the dev guide and link from https://dev.qgroundcontrol.com/en/#translations

bys1123, 2018-09-15:
25% proofreading finished. Sep. 15th

hamishwillee, 2018-09-16:
@dogmaphobic Note, some good movement on other languages too:
[image]