mavlink/qgroundcontrol #3968: WiFi Bridge component not showing up

dogmaphobic, 2016-08-22:
I haven't used the (ESP8266) WiFi Bridge in a while. It was brought to my attention that it is not showing up under Vehicle Setup in QGC 3.0 and I just confirmed it to be the case. Without it, you cannot change its settings. SSID and Password among others.

DonLakeFlyer, 2016-08-22:
There was something going on with parameters coming from additional components. Maybe firmware problem if I remember. Does this trigger from a parameter being available.

dogmaphobic, 2016-08-22:
Yes, it responds to `MAVLINK_MSG_ID_PARAM_REQUEST_LIST` if `param.target_component == MAV_COMP_ID_ALL` or `param.target_component == MAV_COMP_ID_UDP_BRIDGE`. It also responds to `MAVLINK_MSG_ID_PARAM_REQUEST_READ`. Same rules as above (`MAV_COMP_ID_ALL ` or `MAV_COMP_ID_UDP_BRIDGE`). I also handles `"_HASH_CHECK"` returning the CRC.

On the WiFi Bridge firmware side, this hasn't changed since I first wrote it.

Note that when you connect to the vehicle using the WiFi Bridge, the *Bridge* receives all traffic and routes it. It receives the messages from QGC/Vehicle and either passes it transparently to Vehicle/QGC or parses (and responds) the message when needed.

dogmaphobic, 2016-08-22:
The WiFi Bridge component shows up (or not) based on the existence of a parameter:
https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/PX4AutoPilotPlugin.cc#L100-L105

NaterGator, 2016-08-23:
I'm having issues with the RGB LED parameter not showing up until it is changed from the default value too. Maybe related?

DonLakeFlyer, 2016-09-02:
Some details in #1907

DonLakeFlyer, 2016-09-29:
@dogmaphobic Is it supposed to show up when connected over USB? Over usb the additional component is never sent from the vehicle. No QGC problem. Possible firmware problem?

It works fine over WiFi connection. I tried 3.0.0, 3.0.1 and daily. All worked fine.

dogmaphobic, 2016-09-29:
The original issue has been resolved. I forgot to close this issue. No, the WiFi Bridge component will never show up if you connect using USB. There is no "link" between it and the flight controller. It only bridges QGC to it. There is no protocol for a communication from the flight controller targeted directly to the WiFi Bridge. As far as the FC is concerned, the WiFi bridge is transparent.

closed 2016-09-29 by DonLakeFlyer