mavlink/qgroundcontrol #3971: Survey: Camera support discussion

DonLakeFlyer, 2016-08-23:
Current implementation is show below:
[image]
The camera fields are basically a caluclator which calculates a new grid spacing and camera trigger distance each time you change a camera field.

Let use this item to discuss any further feature work here.

kellysteich, 2016-08-26:
We have some additional features that would enhance the camera survey proposed in #3981 .

One feature is being able to choose between several predefined camera models.
At the moment those are hardcoded, but for the future as a more long term solution, we would image that the user should be able to more easily add new camera models, f.ex by defining them in json files, which are then loaded automatically.
For the moment, it would be great to know what camera models others are using, then we could add them to the hardcoded list.

jcooper21, 2016-09-02:
Here is some cameras we are using some with IR triggers and some USB.  Nex 5 (PWM IR trigger), Sony A5000, Sony WX500, Sony  Sony A6300, Ricoh GR2

Otusbubo, 2017-10-06:
We're using the Ricoh GR II and MicaSense RedEdge.  Happy to provide specs on those.

DonLakeFlyer, 2017-10-06:
@Otusbubo If you can provide specs (all the values in custom camera setup) I can add them as standard cameras. Create a separate issue and put the specs in there.

closed 2017-12-29 by DonLakeFlyer