mavlink/qgroundcontrol #3974: Artificial horizon is half black when in vertical climb or dive

kd0aij, 2016-08-24:
[image]

DonLakeFlyer, 2016-11-30:
Mark, thinking about what this should show is making my head hurt a little. The black part should be blue right?

NaterGator, 2016-11-30:
It should be whatever color the other half is. If you're (unfortunately) pointed straight at the ground the two hemispheres in your view are both ground.

DonLakeFlyer, 2016-11-30:
Ah, yeah that's it thanks. Couldn't wrap my head around it as I watched it spin around :)

kd0aij, 2016-11-30:
yes, it's as though you're at the center of a globe with the top hemisphere
blue, and the bottom one green.

DonLakeFlyer, 2016-12-01:
@dogmaphobic Took a quick look at code. But I bet you could fix this in about a minute, whereas it would take me a while to figure out how the code works.

closed 2017-01-03 by DonLakeFlyer