mavlink/qgroundcontrol #3977: Log Download may not be downloading large files correctly

DonLakeFlyer, 2016-08-24:
I was pulling ArduPilot log files from the Solo from some Survey test runs. When I loaded them in APM2 to look for the camera messages they weren't there. Couldn't figure out why. Then I redownloaded the log using APM2 and suddenly the camera log entries showed up. Have a feeling there is a problem with the log download code somewhere which is corrupting the file somehow. Will need to do some for side by side testing/downloads/compares to validate.

NaterGator, 2016-08-24:
I just pulled a 400MB log off through log downloader and verified integrity with a SHA256 hash. That is usually how I test the log downloader code; put a large file with a known SHA256 hash on microsd and download it. 

Any chance you kept the original (corrupt) file?

DonLakeFlyer, 2016-08-24:
I have both versions of the downloaded files. How can I generate the hash for each one?

NaterGator, 2016-08-24:
I use [hashcheck](http://code.kliu.org/hashcheck/) on windows. If you can pull the microsd and get ground truth that is ideal. It would be useful to know if a whole chunk was skipped, just a bin, etc.

DonLakeFlyer, 2016-08-24:
Files are different md5 hash doesn't match.

NaterGator, 2016-08-26:
I was able to repro. A single log_data packet was just missing and all data was shifted to the left by lod_data_payload_len bytes. Looking into cause.

DonLakeFlyer, 2016-08-26:
I'll be hammering on the unit test as well. I want to be able to get the UT to hit the problem before it gets fixed.

DonLakeFlyer, 2016-08-26:
What I have for simulated errors:
* No response to request
* Stream stops before all data sent
* Lost packet in middle of stream
* Corrupted packet

And then the file comparer can determine:
* Missing packet
* Corrupt packet

@NaterGator I wouldn't spend a ton of time debugging this manually. My hope is to figure this out automatically with a good unit test.

DonLakeFlyer, 2016-08-26:
* Dropping random packets doesn't hit the problem
* No response to first chunk request just hangs download (bug needs fixing, but not the corruption problem)

DonLakeFlyer, 2016-08-26:
> I was able to repro. A single log_data packet was just missing and all data was shifted to the left by lod_data_payload_len bytes. Looking into cause.

@NaterGator Do you still have the corrupted file? How does the position of the shift relate to intra-chunk positioning? Looking for a possible boundary case against position within chunk.

DonLakeFlyer, 2016-08-26:
Also in which chunk? First, last, middle?

DonLakeFlyer, 2016-08-27:
In my corrupted file at the point where the files differ from there on out the file contents are all 0's.

DonLakeFlyer, 2016-08-27:
I can't get a repro of this any more, sigh...

NaterGator, 2016-08-28:
@DonLakeFlyer it is on the boundary of a LOG_DATA payload number of bytes, but not an even multiple  of chunksize bytes (somewhere in the early-middle of a chunk).

The chunk was around the 150MB mark of the file (out of ~190MB).

DonLakeFlyer, 2016-12-30:
I can't repro this any more. Moving to 3.2 for some further checking.

booo, 2022-11-01:
We downloaded several GB of logs in the past. Download works in newer versions.

closed 2022-11-01 by booo