mavlink/qgroundcontrol #3980: Stop sending MANUAL_CONTROL messages when joystick is unplugged.

jaxxzer, 2016-08-26:
When the joystick is enabled, if the joystick is unplugged, qgc continues to send manual control messages with the last input received before the joystick was unplugged.

Joystick input does not pick up again when the joystick is plugged back in. The only way to regain control of the vehicle after the joystick is unplugged is to restart qgc.

DonLakeFlyer, 2016-08-26:
@rjehangir Something that will get better with SDL v2?

rjehangir, 2016-08-26:
Yes. Currently SDL1.2 is used on QGC and cannot detect when a device is removed, so there's no good way to do this.

DonLakeFlyer, 2016-10-18:
@rjehangir Fixable now?

rjehangir, 2016-10-18:
@DonLakeFlyer Yes, it should be fixable. This will be taken care of when the hotplugging feature is used. I've got branch I'm working on for that.

closed 2017-02-15 by DonLakeFlyer