mavlink/qgroundcontrol #3982: UVC Video support

cda725, 2016-08-27:
In an effort to expand the capabilities of Qgroundcontrol I am attempting to add UVC video support.

The purpose for this is to use the system as  ground control station and portable ground control station supporting the myriad of wireless video transmitter/receiver systems that are ubiquitous on remote controlled vehicles today. Most are composite or HDMI on the receiver and UVC devices are available with various input options to meet the users needs.

In addition, UVC does not have the processor demand that devices such as the EZCAP and similar devices do nor does it require drivers. Regardless of OS is just "works" wit whatever system you plug it into.

Most DIY (and many commercial) systems make use of the following components;

1. A remote for direct human control
2. A wireless video link (in varying frequencies) to send the video feed from a camera to a receiver
3. Optionally, often  a tablet or portable PC to run software to display telemetry data such as a map overlay, GPS position, speed and so on. On most commercial systems, in addition to the map data and other telemetry information, the APP usually allows a view of the camera feed for FPV operation.

DJI is a great example of a system that uses a remote in conjunction with a tablet. You can use the tablet to display your location on a map overlay, get data about the vehicle, set waypoints and such and you can view the video feed. This simplifies the overall setup and allows a lightweight portable ground control station.

The DJI system is of course proprietary but basically works the same as a DIY model in that there is a transmitter, receiver and computing device on the ground side. 

Qgroundcontrol is a very nice system and would allow someone to easily clone the commercial systems on a budget. Adding UVC video support would allow several configurations. I will give an example of one below:

Assuming a UAV ( quad drone) build for our example the following would be installed:

1. Gimbal(s) 
2. Camera(s)
3. Wireless telemetry module
4. Pixhawk
5. Wireless video transmitter (which are small, light, powerful and will most will run from 3S to 7S
6. Remote receiver such as an X8R

On the user end the following in our example is used:

1. Taranis X9D remote
2. Android tablet
3. USB hub
4 UVC capture device for the video receiver
5. USB Wireless telemetry receiver.
6. Video receiver with composite or HDMI out

I have tested the USB wireless telemetry modules and can fly several miles with my remote and Qgroundcontrol running from my Android and Windows tablets.  I currently have to run a separate monitor to view my cameras. If one did not need the separate monitor for this configuration they could mount the components to the back of the tablet stand that is attached to the remote and power everything from the Taranis (or whichever brand remote they choose).

QGC has a nice video window feature and if it allowed a USB capture device, specifically driverless UVC capture devices (since again UVC will work in Android, Unix, Linux, OS X and various other operating system) so that a wireless receiver can be plugged in to it, this would make QGC a much more powerful and versatile package.

I am not sure I understand the original intent of the video system implemented into the QGC system as it appears to require a lot of gear to get an image to QGC that could be simplified with UVC. With wireless long-range telemetry and video systems being inexpensive and readily available, in addition to weight in flying machines being a serious consideration  it appears like a logical transition for QGC and a much necessary implementation.

To be able to use QGC as an "all-in-one" app would exponentially expand the possibilities of QGC and its uses. the 3DR Tower app has now made this a standard feature so the industry is certainly leaning toward versatility regarding video. QGC in my opinion is a better app, has better response and the layout is much more appealing. Not to mention that Tower appears to have a latency issue with wireless telemetry devices where QGC does not.

I have written a simple QT camera app that will look for a UVC device (since most tablets have built-in cameras)  in QT 5 (the same version used to create and build QGC) but the Gstreamer is so heavily nested into QGC I lack the time to reverse engineer its implementation and add the simple UVC plugin. I have also tried to modify the Gstreamer implementation to use UVC but I don't believe this is possible in a method that would be stable - perhaps I am wrong here but after a few attempts and researching the matter it looks like using the built-in QT features is a better method.

I propose an implementation of USB UVC as outlined above and hope there is general interest in the QGC community to prioritize said implementation.

Thank you to the QGC community and my fellow developers and enthusiasts for reading this issue and considering the proposed implementation.

dogmaphobic, 2016-08-27:
Technically, it's two lines of code. Literally. A few more for configuration (selecting which camera in case you have more than one UVC device). I can have a desktop version done by tomorrow (not in front of a real computer right now). I don't have anything that I could plug into my tablet/phone to test and there isn't anything that would work with iOS.

cda725, 2016-08-27:
Thank you for your reply. I wish I knew the code in QGC as well as you. If I could get a desktop version I could probably get it working on android. I would be forever indebted to you if you could make the time.

Regarding iOs, there are actually wireless telemetry systems and UVC devices that will indeed work. I am an Apple developer and have tested my UVC device and it does work. I have not ordered and tested the telemetry system yet but I am sure it works. See the following links:

http://www.ebay.com/itm/UVC-Android-Video-Capture-Card-Android-Linux-iOS-Windows-usb-Capture-Card-/161108508329

This is basically what I have but made by "MINE" - works fine in iOS

and

https://www.amazon.ca/USB2-0-Capture-Supports-Android-Windows/dp/B00NIEQQ4Y

and for telemetry there are devices for older iPhones and iPads and I know of a company working on a 915Mhz model now.

Here is a link for one that I don't believe is mavlink but it shows there are devices coming out.

http://www.goodluckbuy.com/jiyi-p2pro-100mw-wireless-telemetry-module-ground-station-433mhz-915mhz-for-ios.html

Kind regards,

Gio

cda725, 2016-08-28:
@dogmaphobic 

So can I enlist you in helping? Once I know what to change I need to add a switch to allow a user to switch between the current video source solution and the new UVC solution. 

I truly respect peoples time and if you have the time to code out the desktop version or at least point me to what needs changed I would really appreciate it.

Kind regards,

Gio

dogmaphobic, 2016-08-28:
I'm already on it. USB video is already working. I am now adding configuration so the user can set what video source to use. As long as the video source shows up as a "camera", it will be available for use.

cda725, 2016-08-28:
@dogmaphobic 

Gus,

You are awesome. I had hoped to contribute but your FAR more capable,  experienced with the QT environment and QGC than I am. 

Do you plan on creating a branch on github? If so I will start on the Android (and iOS side) immediately! Otherwise email is fine too...

dogmaphobic, 2016-08-28:
Yeah, I figured it would take anybody else a whole lot longer to get around QGC's architecture. The code is the same for all platforms. If a video source is available as a "camera", it will show up as an option for "Video Source". I'm almost all done. I'm now working on making the video controller a singleton so it can be accessed from both the configuration panel as well as the video view (in order for the settings to affect the one and only video controller). Once done, I will push to master and we can start experimenting with USB video capture devices attached to tablets (it already works with cameras, either builtin or attached to the USB).

cda725, 2016-08-28:
@dogmaphobic 

That's great! Hopefully I can be of some help. Possibly more on the iOS side perhaps... I can't wait to tweak around and try it on Android right away. 

If you can, please let me know what you changed (briefly) and I will do a "Diff" so I can better understand the structure and perhaps be more helpful on either this or future endeavors.

I have LOTS of hardware and many devices - my job gives me access to quit a bit so I can test a broad range of devices and configurations.

Thank you again for doing this. I believe it will really increase the functionality and usability of an already great program!

Let me know how I can help...

Kind regards,

Gio

dogmaphobic, 2016-08-28:
@cda725 Check #3983 for the diff. PR is ready. Once it passes CI, I can merge it so you can experiment with whatever devices you have. We can take it from there.

cda725, 2016-08-28:
@dogmaphobic 

Damn you are fast. This is great. I can truly say I am excited not only for myself but the many of us who have wanted for this so badly. 

Thank you again. Let me know what I can do!

dogmaphobic, 2016-08-28:
@cda725 It's ready to experiment with. Use `--logging:VideoManagerLog` to see some debug info from the video manager. It will list all *cameras* found (though you can see them in the combobox within General Preferences as well). What I'm curious is to see if these devices show up as *Camera* to the OS. Otherwise this won't work as is.

cda725, 2016-08-28:
@dogmaphobic 

Awesome! I am on it now!!!

cda725, 2016-08-28:
@dogmaphobic 

Awesome work!!! We are close. 

Regarding the Android,  I have some code that will look for UVC as a camera that is working in a self contained app. With your mods  I can see the front and rear camera and a USB web camera but UVC devices are not showing up. I believe I can fix this by adding my code to yours in the way it is detecting "cameras".

Also I may have found the root of the camera switch bug which I will address with my code modification. I don't want to speculate until I have tested locally.

I am undoubtedly not got to be as quick as you but I will get on it - you are obviously a far more proficient programmer than I and I have great respect for that!

I am going to try the desktop Windows version and OS X version as well to see if UVC is working there since I know that OS X is treating it as a camera.

Oddly, Camera-Fi treats it as a camera (of sorts). I am just going to apply a patch similar to theirs for the UVC.

dogmaphobic, 2016-08-29:
@cda725 In my drawer of old junk, I found an USB video capture dongle but it doesn't work. At least not on Mac OS. It's listed as:
```
SM-USB 007:

Product ID: 0x0007
Vendor ID: 0x1c88  (Somagic, Inc.)
Version: 1.00
Serial Number: SMBL007
Speed: Up to 480 Mb/sec
Manufacturer: Somagic, Inc.
Location ID: 0xfd313000 / 4
Current Available (mA): 1000
Current Required (mA): 200
Extra Operating Current (mA): 0
```

Do you know of something I could order to test this? Something that does show up as a device in `QList<QCameraInfo> cameras = QCameraInfo::availableCameras();`?

dogmaphobic, 2016-08-29:
Same for an old Hauppauge component video capture device, which I didn't exactly expect to work but what the heck :)

```
Hauppauge HD PVR:

Product ID: 0x4903
Vendor ID: 0x2040  (Hauppauge Computer Works, Inc.)
Version: 0.00
Serial Number: 00A4FC0A
Speed: Up to 480 Mb/sec
Manufacturer: AMBA
Location ID: 0xfd313000 / 4
Current Available (mA): 1000
Current Required (mA): 4
Extra Operating Current (mA): 0
```

cda725, 2016-08-29:
@dagoodma 

A simple capture card wont work as easily due to driver issues. A UVC is driverless. QT should enumerate a UVC device (newer versions of QT do) but this version is not. A VUC device can be had for 10.00 but is still going to require my UVC patch that I am working on now. 

Also, I believe your camera switch bug is from not killing the stream. on change then restarting it. I haven't looked through all 14 files you changed but I believe this is the switch issue.

I noticed in writing my little stand alone app for UVC in QT that if I didn't stop the camera prior to a switch it would hang on whatever the last camera I chose was.

Let me know if you want a UVC device - I would be happy to order you one.

Kind regards, 

Gio

If you are in the states I can order one from Amazon for you.

dogmaphobic, 2016-08-29:
NC. As *states* as it gets. Do you have a link?

cda725, 2016-08-29:
Yes - https://www.amazon.com/s/ref=nb_sb_noss_2?url=search-alias%3Daps&field-keywords=UVC+capture+card

dogmaphobic, 2016-08-29:
Yes, I'm not "stopping" the camera on switch. Just changing the `deviceId`. I set the camera mode to *ViewFinder* (as opposed to Picture or Video capture).

cda725, 2016-08-29:
I am in Virginia and Texas (Texas at the moment)

dogmaphobic, 2016-08-29:
The one I found is almost exactly like this one. Mine has an additional S-Video plug: https://www.amazon.com/Generic-Usb2-0-Capture-Compatible-Window/dp/B00RMYWGWC/ref=sr_1_1?ie=UTF8&qid=1472430241&sr=8-1&keywords=UVC+capture+card

cda725, 2016-08-29:
EZcap is confusing - most are generic but you will be OK as long as it is UVC... I have this one:

https://www.amazon.com/Vktech-Professional-Quality-Android-Capture/dp/B00JR06TZO/ref=sr_1_11?ie=UTF8&qid=1472430193&sr=8-11&keywords=UVC+capture+card

cda725, 2016-08-29:
That is what I mean by EzCap being confusing. Most are built on a chipset needing a driver. Only a few are UVC. The one you found is UVC so it is a different beast altogether. Standard capture cards/dongles are a PITA - UVC is much easier. To get a standard EzCap to run on an android for example you would need to be rooted and build a custom kernel

dogmaphobic, 2016-08-29:
I will wait until you have something that works. I'm not going to *experiment* with a $50 device that I would otherwise have no use for :P

cda725, 2016-08-29:
Like I said... I would be happy to send you one. The 10.00 and 50.00 are pretty much the same as long as they are UVC. UVC and standard capture devices are worlds apart.

Let me know - I have an expense account and can easily write off 10 to 20.00 for a card for you.

With that said, I should have something late tonight or tomorrow. Also going to try to patch the toggle to kill the current stream prior to making a camera switch. This should fix the camera hang issue (I hope - lol)

cda725, 2016-08-29:
@dogmaphobic 

Got the UVC working in Windows!!!

cda725, 2016-08-29:
Works great! - This is progress. A 10" windows tablet is as cheap as an 8" android (if you want any power that is)

dogmaphobic, 2016-08-29:
Cool. I have an Immersion AV receiver and all my old drones have analog FPV video. I've been in the digital side for so long now that it never occurred to me how useful this could be. If nothing else, I can hookup some goggles to a tablet and have an extremely glorified QGC OSD :)

cda725, 2016-08-29:
It has SOOOO many uses. If you want, shoot me an email and I will give you my number - the least I can do is get you a device to help me refine this for the sake of the whole community; especially since you did most of the work.

Plus I want to pick your brain about the digital setup you have.

 g.michell (at) cdagency (dot) us

I haven't gone digital yet due to the fact that we have so much analog gear from expensive high-res telephoto cameras, thermal imagers and so on that have simple composite out - inexpensive to just hookup a high-power analog transmitter/receiver and camera switcher with 2 or 3 gimbals (we also have digital night vision on one of ours - it is pretty cool) . I will email you a link of a test surveillance from 900' - I think you will enjoy the thermography considering we were in pitch-black conditions...

cda725, 2016-08-29:
I just sent you a 20.00 gift card from Amazon to your @grubba.com email. Now you can pickup a UVC device :)

cda725, 2016-08-29:
@dogmaphobic 

Please confirm you receive it. Should be less than 5 minutes.

cda725, 2016-08-29:
@dogmaphobic 

I hope all is well and you got the amazon thing...

In my app I have a function that basically reads: Qcamera::Unload();
This is required to switch between cameras using the chosen method.
In needs to be in the VideoManager.cc file under VideoManager starting at line 109.

I am going to implement it somewhere around line 117. I have to do it different than my standalone app since I actually use the term QCamera in my app. This will fix the video switch issue.

Regarding UVC not working in Android I have a workaround. I will note that your work allows not only for UVC devices, but if you are in windows for example, you can use many standard USB capture cards as long as the driver is installed. Although they are not cameras they are enumerated as such so this is VERY good.

Just wanted to let you know for sure what the deal is with multiple USB devices.

For reference see: 

http://lists.qt-project.org/pipermail/interest/2014-January/010475.html

The pipeline has to be closed before selecting a new one.

cda725, 2016-08-29:
@dogmaphobic - I fixed it - Yeah!!! No more USB bug! you can now switch devices on the fly!

Just add one one of code at line 117 of VideoManager.cc

closed 2016-09-18 by cda725