mavlink/qgroundcontrol #3984: Astyle formatting

dagar, 2016-08-28:
As requested https://github.com/PX4/Firmware/pull/5370

I propose we duplicate the PX4 Firmware setup. Same astyle options, helper scripts, and travis-ci enforcement.

NaterGator, 2016-08-29:
If Qt Creator is going to continue to be the primary IDE it will fight with astyle with default settings. We should probably distribute a better code-style XML file for Qt Creator so the default codegen helpers don't create too much drama with astyle. We can manipulate and export one in Tools->Options->C++->Code Style

DonLakeFlyer, 2016-08-29:
> Same astyle options,

By that do you mean same coding style? If that's the case I would disagree. I want to keep the existing coding style for QGC.

dagar, 2016-08-29:
My main interest is letting a tool handle it so we never have to think about it again. For the most part I don't care about the specifics as long as it's consistent. 

@dogmaphobic was interested in sharing the same style between PX4 and QGC to make it easier to go back and forth.

dogmaphobic, 2016-08-30:
The main (and possibly only) change would be the use of tabs for indentation. It should be one single change in Qt Creator/Sublime Text/WordStar/Whatever. It is a bit of a pain to remember to switch each time you go from one project to the other.

DonLakeFlyer, 2016-08-30:
Right now I think the coding style for QGC is pretty close to the QtCreator default. I don't remember changing anything there. Anyone know of differences? I think that's the best to go with since it requires the least setup for most people just getting into QGC dev.

NaterGator, 2016-08-30:
It is pretty close, I think we just need to specify things like preferred indentation style. Once it settles I'll create an XML file we can post on the wiki for people who want to exactly match the style without having to think about it.

dagar, 2016-08-30:
There's also the Beautifier plugin that will directly use astyle within QtCreator.
Help > About Plugins > C++ > Beautifier

NaterGator, 2016-08-30:
That's true too. The PITA is QtCreator's default behavior is to automatically adjust whitespace for any line you've edited to conform with your current C++ code style settings. If you replace spaces with tabs but QtCreator is set to use spaces it will undo your change on that line and replace tabs back to spaces. I don't know if it can use beautifier for that step instead of the built in C++ styles, but if so that would be helpful. Otherwise I know this is going to be an ongoing annoyance if we don't get QtCreator's C++ style to match what astyle ultimately wants.

closed 2024-04-09 by HTRamsey