mavlink/qgroundcontrol #3985: QGC - OSX activates always FaceTime camera

lvale, 2016-08-29:
Hi

Even if the Facetime camera is not selected the camera is active (Green LED). With all the privacy mania, I believe the Facetime camera should only be active if explicitly selected

[image]

[image]

dogmaphobic, 2016-08-29:
Yes, the camera should be "stopped" when not using. Still working on it...

lvale, 2016-08-29:
btw: love the camera features on the Android app :) Works great with OnePlus ;)

lvale, 2016-08-29:
@dogmaphobic @DonLakeFlyer As a bonus feature of this camera issue a record video stream function would be nice. Next would be to have data overlayed on the video, and recordable :)

dogmaphobic, 2016-08-30:
> As a bonus feature of this camera issue a record video stream

Record the stream is not simple either but not impossible either. Recording the composite (video and overlay) is not simple at all. On desktop, you're better off using QuickTime or something like it. On mobile, no clue. I like the idea though...

dogmaphobic, 2016-08-30:
Fixed in #3986

closed 2016-08-30 by dogmaphobic