mavlink/qgroundcontrol #3987: Deleting inserted waypoints bug

kellysteich, 2016-08-29:
When planning a mission, inserting a waypoint by selecting another waypoint and choosing "Insert" from its menu and then deleting the inserted waypoint breaks the mission planner view, as seen in the following screenshots:

mission in beginning           |  mission after inserting a waypoint |  mission after deleting said waypoint
:-------------------------:|:-------------------------:|:-------------------------:
[image]  |  [image] | [image]

To reproduce:
* create a few waypoints normally, by clicking inside the map
* then insert a waypoint, by selecting one of the previously created waypoints and choosing "Insert" from its menu
* delete this inserted waypoint, by choosing "Delete" from its menu

Saving a mission with a waypoint that has been inserted and reloading it solves the problem and the waypoint can then be deleted normally.

birchera, 2016-08-31:
We did some research on this but could not quite put our finger on the right spot. We made the following observations:

* It only occurs when adding items in the middle of the list and not appending them. The function [insert(i, object)](https://github.com/mavlink/qgroundcontrol/blob/d937bd2b6ac4048a5872cd5562fdd614ab3fb8f1/src/QmlControls/QmlObjectListModel.cc#L171) seems to call [different methods](http://doc.qt.io/qt-4.8/qlist.html#insert) Qt internally depending on the value of i.
* We could not reproduce it when setting a breakpoint in the function [QmlObjectsListModel::removeAt](https://github.com/mavlink/qgroundcontrol/blob/d937bd2b6ac4048a5872cd5562fdd614ab3fb8f1/src/QmlControls/QmlObjectListModel.cc#L138).

DonLakeFlyer, 2016-08-31:
It is most likely a Qt draw bug. I've hit this before with insert/delete. The trick is to find a workaround that works. Need to try it on a mobile build to see if it a QQuickWidget problem versus native Qml. Haven't had time yet.

DonLakeFlyer, 2016-09-04:
This repros on mobile native qml QGC as well

DonLakeFlyer, 2016-10-20:
Testing for fix in Qt 6.2

DonLakeFlyer, 2016-12-28:
Fixed in 5.8

DonLakeFlyer, 2016-12-28:
Will be fixed in 3.2 when we move to Qt 5.8

DonLakeFlyer, 2017-03-19:
Fixed now that we are on Qt 5.7

closed 2017-03-19 by DonLakeFlyer