mavlink/qgroundcontrol #3988: QGC - Video stream recorder option

lvale, 2016-08-29:
A great addition on the video stream would be to have the possibility of recording the video stream, preferably with a Data overlay (like a OSD, but on QGC)

dogmaphobic, 2016-08-30:
Repeating here as I had not noticed you had opened an issue for it:

Record the stream is not simple but not impossible either. Recording the composite (video and overlay) is not simple at all. On desktop, you're better off using QuickTime or something like it. On mobile, no clue. I like the idea though...

lvale, 2016-08-30:
@dogmaphobic Thanks. The Quicktime option, which I use extensively, I believe is only available on Macs. Lazy me that I didn't check recently, but I believe there's an option on gstreamer to save a stream to file, but I don't know how it is now.

QuiGon2, 2017-02-02:
Hi all,

I have been using 3rd party windows apps for recording the video, so I'm so happy for having this option in QGC. Is it possible to choose other formats for the output video (.avi, .mp4...)? Windows player does not play the output mkv I got this morning... I thing it is also necessary to overlay the navigation data (voltage, depth (ArduSub), and so)

DonLakeFlyer, 2017-02-07:
@dogmaphobic Do you know about this with the new record video stuff?

DonLakeFlyer, 2017-04-06:
Available now

closed 2017-04-06 by DonLakeFlyer