mavlink/qgroundcontrol #3993: Survey: trigger calculations after change

AndKe, 2016-08-30:
After changing altitude of a survey area grid, grid spacing and trigger distance is not recalculated.
Please make sure changing values triggers update.

DonLakeFlyer, 2016-09-04:
It doesn't quite work that way. Right now if you change altitude it leaves spacing/trigger alone and recalcs GSD and overlaps values. Not quite sure what is really the right thing here.

@birchera @kellysteich FYI for discussion

Also I think we may need an option to turn off GSD as a setting.

AndKe, 2016-09-04:
I'd say, the most intuitive, would be to always recalculate things by last
(manually)modified setting, anything else would require knowledge of the
inner workings, and allow mistakes.

For example, as it is now, user may plan a grid, at a given side ovrlap,
then decrease altitude to increase GSD, and thus get too little side
overlap as grid spacing is not recalculated on altitude change.(and user is
unaware of that)

birchera, 2016-09-05:
The underlying idea of the way we connected these values is the following:

* Previous way of working is maintained by allowing to set up everything from altitude, grid spacing and triggering distance. When setting up the mission this way, GSD and overlaps are more a kind of informative display.
* Overlaps and GSD are handled as a more high level layer of planning. When changing one of these, altitude, spacing and trig. distance are adapted accordingly.

While the former allows to specify the path, the latter option rather enables the planning starting from measurement requirements.

DonLakeFlyer, 2016-09-30:
If you look at the pull you'll see the ui for this now. The key issues are between Altitude and Ground resolution. You can now pick which one of these you want to take precedence. All values will recalc based on that.

closed 2016-09-30 by DonLakeFlyer

AndKe, 2016-09-30:
it looks *very* , intuitive, display-space-saving , and efficient.  great solution.

DonLakeFlyer, 2016-09-30:
Thanks I was hoping I wasn't the only one that understood it! Really trying hard to improve usability.

AndKe, 2016-10-01:
Now all mission items are trimmed like this: (tested on two Ubuntu systems)
[image]

this anomaly is not in QGC settings, parameters, or any other place than mission items.

AndKe, 2016-10-01:
it's most likely caused by all the "height: 1" lines, like 
(line 278)
https://github.com/mavlink/qgroundcontrol/pull/4087/files#diff-4c77598413c23d8381a79cb4086c69f9R278

please observe the previous, good definition...

DonLakeFlyer, 2016-10-01:
Very strange. Can you move to a separate issue and I'll take a look. The hieght 1 rectangles are dividing lines. Nothing to do with the inner part of text fields.