mavlink/qgroundcontrol #3998: Missing runtime dependency libQt5MultimediaQuick

birchera, 2016-08-31:
Unable to deploy on Ubuntu 14.04.

When running my own precompiled qgroundcontrol, it only displays a white screen and the command line tells:

> qrc:qml/MainWindowInner.qml:283:5: Type FlightDisplayView unavailable
> qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayView.qml:17:1: plugin cannot be loaded for module "QtMultimedia": Cannot load library /home/wingtra/qgroundcontrol/release/Qt/qml/QtMultimedia/libdeclarative_multimedia.so: (libQt5MultimediaQuick_p.so.5: cannot open shared object file: No such file or directory)

I assume it has something to do with #3983 that brought in the multimedia dependency.

This also happens with the daily build of yesterday. Android tablet works fine.

DonLakeFlyer, 2016-08-31:
I think @dogmaphobic may have just fixed this

dogmaphobic, 2016-08-31:
Not yet. I had added `libQt5Multimedia.so.5` but I missed `libQt5MultimediaQuick_p.so.5`. It will be a trial-and-error until we get it all there as I don't have a way to fully test what libraries are needed. I'm about to make a commit with `libQt5MultimediaQuick_p.so.5`. Let's see if this is all it takes.

dogmaphobic, 2016-08-31:
It's in #4000

kellysteich, 2016-09-01:
Thanks a lot for the new fix in #4000 , this works now for us!

closed 2016-09-02 by birchera