mavlink/qgroundcontrol #4007: Adding SRTM 30m and 90m to terrain altitude

kikislater, 2016-09-01:
Hi,

With Elevation here : http://elevation.bopen.eu/en/stable/quickstart.html it's possible to simply get SRTM 1 @30meters resolution and SRTM3 @90meters resolution.
What do you think about implement it in QGC. Note that this tool has a dependency which is gdal and could be very usefull to add some more features later in QGC. Gdal/ogr is compatible with multiple platforms like ios, android, win, unix.

DonLakeFlyer, 2016-09-01:
At some point we will add height over terrain support. When we get to that we'll consider using this.

Antiheavy, 2018-11-24:
@kikislater QGC now supports terrain database information (from Airmaps I believe) for mission planning.  Is this issue still applicable?

kikislater, 2018-11-26:
I don't think airmap have 30m srtm but you can close !
May be a better workaround would be to load our own elevation data.

closed 2026-10-04 by github-actions[bot]