mavlink/qgroundcontrol #4008: Feet to metric no round tripping correctly

DonLakeFlyer, 2016-09-01:
Type in 100 ft to GF_MAX_HOR_DIST and it will roundtrip back to 98.4521... feet!

dogmaphobic, 2016-09-01:
Imperial feet went up the Matterhorn and got frostbitten...

NaterGator, 2016-09-01:
This again?! Is something in the path bypassing the fact translators?

NaterGator, 2016-09-01:
Oh. Here's why:
https://github.com/mavlink/qgroundcontrol/blob/801fdde8e57a04a4b344621cf3809c7b968166af/src/FirmwarePlugin/PX4/PX4ParameterFactMetaData.xml#L2216

100 feet to meters = 30.48m
((int)30.48) meters to feet = 98.4521ft

DonLakeFlyer, 2016-09-02:
Whoopsie

DonLakeFlyer, 2016-09-04:
There is still a problem with the translators not converting back to int for these params.

DonLakeFlyer, 2016-09-04:
This is fixed now. Translators are only installed in facts which are real numbers.

closed 2016-09-04 by DonLakeFlyer