mavlink/qgroundcontrol #4011: Arrows to indicate flight direction in mission planning mode

kellysteich, 2016-09-02:
We are currently working on a feature that adds direction arrows to the lines connecting the way points in the mission planning mode in order to better visualize the flight direction.
It look like this right now:
[image]
The plan is to add the same arrow indicators to the survey grid.

We were wondering if there is any interest in us pushing this upstream for the rest of the community?

Also currently we are loading the image of an arrow and adding this to the waypoint lines,does anybody maybe know of a better way of doing this, f.ex something more along the line of a qml object like the MapPolyline used for the waypoint lines?

DonLakeFlyer, 2016-09-02:
Using an image is probably fine. Other way would be to create your own custom control using Qml Canvas and then managing the positioning yourself.

Antiheavy, 2018-02-04:
+1 for direction arrows in Upstream

booo, 2022-11-01:
Direction arrows are implemented.

closed 2022-11-01 by booo