mavlink/qgroundcontrol #4017: APM/PX4 Camera trigger parameters

jcooper21, 2016-09-04:
@DonLakeFlyer For trigging cameras in APM these setting are needed in QGC to setup the camera.
[image]

dogmaphobic, 2016-09-04:
The current UI is mostly still based on using the pins as digital IO (Trigger Interface type *GPIO*).

I think what would make sense is to have the lower portion of the UI to change based on the current *Trigger Intertace* type. The existing UI is for GPIO and when you change it to PWM, the extra parameters should replace the pin mapping.

[image]

Mavlink has a message to handle active duration (`TRIG_ACT_TIME`) but it's not currently present in the UI. It's probably a good idea to bring that up for both modes. PWM limits however, are not present in the TRIG_XXX commands. That would have to be added as far as I know.

What is AC-T mode? These modes (AC-T, SS-T, TL-T) should be spelled out as it may not be obvious to most people.

I can put this all together quickly once the PWM values issue is resolved.

What about trigger feedback like you had described earlier? Where would you define the GPIO pin to sense it?

P.S. Gimbal control is another matter altogether and deserves its own, separate page. Like camera trigger, there are different ways to interface with gimbal controllers. PWM (as pictured above in the MP UI) and serial protocol (Mavlink or Proprietary like the SimpleBGC).

Camera control is also a lot more than just this. Here we are talking about just a simple binary state. There is also an entire set of camera settings that will need to be addressed for when you are using the camera for other functions other than mapping.

DonLakeFlyer, 2016-09-04:
This is ArduPilot only for now since PX4 firmware doens't have support for it

jcooper21, 2016-09-05:
Off the top of my head I believe for feedback we just select which pin in the parameters for APM

jcooper21, 2016-09-12:
@DonLakeFlyer @dogmaphobic Looks like were going to have the same functionality in PX4 this week.  Is there any chance we can get those fields in QGC so I can test out the PWM trigger @dagar is doing this week.

https://github.com/PX4/Firmware/issues/5437#issuecomment-244580013

dogmaphobic, 2016-09-12:
Once the new MavLink messages are in, it shouldn't be too hard to add. Regardless, I will need to re-work the UI. The GPIO pin numbers should not be displayed if the trigger interface selected is the Seagull. Technically, that option should not even be there until we can actually support it.

jcooper21, 2016-09-12:
I think that may be backwards for example the pin number should be there always, so we can select what pin we have our trigger cable in (and eventually the feedback cable).  That part that does not need to be there is the seagull drop down selection.  All we need is the blank fields because the way it is now we cant change modes with seagull cables from IS-T to AC-T trigger modes because there hard coded.  Trigger interface selection should just be PWM or the other mod (I cant remember but its the one you educated me on earlier)

jcooper21, 2016-09-12:
@dogmaphobic I forgot to address your question above about modes such as AC-T.  Those are seagull specific modes for their hardware each company has there own terms for the trigger types such built into their products.  All we care about is the pin selection for the cable, Push, not pushed, interval and servo settings.  With that we can trigger anything.

DonLakeFlyer, 2016-11-24:
I'm going to move this to 3.2. Power users can still setup with parameter editor

DonLakeFlyer, 2017-05-04:
This is going to miss 3.2 as well

DonLakeFlyer, 2017-12-29:
@dogmaphobic This is in there now isn't it with the new camera settings thing?

dogmaphobic, 2018-01-02:
Not really. From what I understand, the stuff above are for "dumb" cameras where you need actual IO pins to trigger/get feedback. The new camera interface/settings are for intelligent camera/controllers that speak MAVLink where messages are sent/received for triggering/receiving feedback (and no IO pins are involved).

DonLakeFlyer, 2018-06-20:
Closing until PX4 has mavlink support

closed 2018-06-20 by DonLakeFlyer