mavlink/qgroundcontrol #4020: ParameterLoader can't handle parameter count changing after initial load

DonLakeFlyer, 2016-09-04:
ArduPilot has the concept of enable params which then cause additional parameters to load on the next refresh. Currently the parameter loader does not update it's param count after initial load which means this won't work correctly.

edwinhayes, 2018-11-12:
@DonLakeFlyer do you know if there was/is any plan for addressing this?  I note that (on AC 3.6.0 at least) if you hit the refresh button to reload ArduPilot parameters, QGC gets stuck eventually asking for parameter whose index is higher than the count field being returned by the flight controller.  I'm not even sure if ArduPilot guarantees that parameters are reported with the same indices on successive reads?

DonLakeFlyer, 2018-11-12:
I'm not sure that problem is the same as what this issue is describing. Can you enter a seperate issue for that and include ParameterManage logs for it.

closed 2026-10-04 by github-actions[bot]