mavlink/qgroundcontrol #4024: PX4 flash all branches

dagar, 2016-09-05:
On the PX4 dev call we've been discussing ways to make it easier to involve testers outside of the development community. One idea is to extend QGC advanced flashing to include all branches on the PX4 repo. 

On the PX4 side I would need to extend the build system to upload all branch builds (and delete as needed), but that's easy. Then in QGC we either resurrect and finish the json build file (https://github.com/mavlink/qgroundcontrol/issues/2797) or query the available branch builds through the github api.

@DonLakeFlyer you have some experience determining with the github api and QGC. Which way would you like to go?

DonLakeFlyer, 2016-09-09:
I'll take a look when I get back home next week. Can't remember what state the current firmware version loading stuff is in. So no comment till then.

DonLakeFlyer, 2016-09-13:
I think we go back to the json file #2797 and finish that implementation.

closed 2026-10-04 by github-actions[bot]