mavlink/qgroundcontrol #4031: absorb mavlink and gpsdrivers submodules

dagar, 2016-09-10:
@LorenzMeier @DonLakeFlyer - If I made it trivial (and automated) to pull in new versions of mavlink and gpsdrivers would either of you be opposed to getting rid of git submodules in this project? 

My primary reason is that launchpad isn't able to handle them, but I think it would also reduce developer pain. Launchpad can automatically build QGC on each of the supported Ubuntu versions and create both stable and nightly development PPAs. It would also be a step towards having QGC officially in Ubuntu if desired.

QGC doesn't have the PX4 submodule helper scripts that make working with submodules less painful.

https://github.com/mavlink/qgroundcontrol/issues/3886
https://github.com/mavlink/qgroundcontrol/issues/665

DonLakeFlyer, 2016-09-13:
Are you talking about just integrating the helper scripts like PX4 or getting rid of submodules all together? I'm not sure getting rid of submodules would be a good idea.

dagar, 2016-09-13:
I'm talking about getting rid of them given it's only mavlink + gpsdrivers. To update mavlink there'd be a helper script to generate mavlink in tree then you'd commit the result along with corresponding qgc for those new messages. 

If you don't like that idea, then alternatively I'd have a readonly QGC mirror with the submodules absorbed that exists just to make launchpad happy.

DonLakeFlyer, 2016-09-13:
Not quite sure what to think about this. Getting rid of submodules just doesn't smell right. Would be interested to here what other people think: @dogmaphobic @LorenzMeier ?

dagar, 2016-09-13:
It would be just like the other libs that have been committed.
https://github.com/mavlink/qgroundcontrol/tree/master/libs

LorenzMeier, 2016-09-14:
I think its inferior in managing dependencies, but having an Ubuntu package would be really important (app image is great, but its not auto-updating). So I'd vote for copies despite being horrible.

DonLakeFlyer, 2016-09-14:
@dagar If you get rid of submodules would it be possible to create some scripts for something so we have a standard way to update what used to be in submodules?

DonLakeFlyer, 2016-09-16:
I'm ok with this if we can create some scripts that will pull the updates.

dagar, 2016-09-17:
Great. I'll do it in a branch off the main repo and make sure everyone is okay with the result before anything changes in master.

closed 2017-06-26 by dagar