mavlink/qgroundcontrol #4041: NO gps lock

anujkumargupta, 2016-09-16:
Hi all i am new to pixhawk and qgc.I calibrated all the sensors but  I have following issues.

1. I am trying to fly in mission mode or position mode . I planned a mission but whenever i try to switch the flight mode to mission from manual it says "set flight mode command temporarily rejected" .

2. I dont have any rc transmitter or receiver so i am using a virtual joystick from qgc and i also set the parameter COM_RC_IN_MODE to joystick/no rc checks but when i tried to arm the vehicle in manual mode it says "armed denied : manual throttle non zero".

3. i also connected the gps module but in flying tab of qgc it says "no gps lock for vehicle"

Please help
Thanks

DonLakeFlyer, 2016-09-16:
No GPS Lock is a vehicle hardware setup problem. QGC is just reporting back what the vehicle is telling it. You'll need to debug with the firmware folks.

For #2 what build version are you using?

DonLakeFlyer, 2016-09-16:
For #2 is the throttle virtual joystick showing as centered or all the way down when you try to arm?

anujkumargupta, 2016-09-19:
Thanks for the reply 
I am using the daily build version of qgc.
My joystick postion is in center

DonLakeFlyer, 2016-09-19:
The throttle stick should start in the down position. It must be down to arm. It should also not recenter itself. If your virtual throttle stick doesn't do that you aren't running the latest version of QGC.

anujkumargupta, 2016-09-20:
Thanks ,that problem is solved using the radio telemetry ,Earlier i was connecting the pixhawk directly with usb so it shows warning but now after arming when i give the command to take off it says takeoff denied , disarm and retry.
And what about gps lock prob how to fix that.
Thanks

DonLakeFlyer, 2016-09-20:
You'll have to ask firmware folks about gps problems. That isn't anything QGC is in control of. So as far as I can tell there is no QGC problem here.

closed 2016-09-20 by DonLakeFlyer