mavlink/qgroundcontrol #4055: Android radio connection issue

dagar, 2016-09-18:
Reported through the play store. He said the connection with this radio works with Tower, and he's tried removing Tower and 3DR services.

Device L-01F (g2)
App version 301000 (3.0.1)
OS Android 5.0

    Error on link. Error connecting Could not create port Unknown error

[image]
[image]
[image]

DonLakeFlyer, 2016-09-19:
Any way to get the person over here? Are they getting the connec to QGC prompt when they plug in?

bestpest, 2016-09-20:
I'm that person. I'm getting the connect prompt when connecting radio.
But qgc can't connect, in the port menu there's no port. Don't know if it's okay.

DonLakeFlyer, 2016-09-20:
@dogmaphobic Gus, will serial ports from Android show up in LinkManagerVerboseLog: https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/LinkManager.cc#L493? That would show if QGC ever sees the new port being added.

dogmaphobic, 2016-09-20:
Yes, they should. I haven't tried that in ages though.

DonLakeFlyer, 2016-09-20:
@bestpest Can  you start QGC without the radio connected. Then go to Settings/Console and turn on LinkManagerVerboseLog. Then plug the radio in and post the console log to us.

bestpest, 2016-09-21:
[qgclog.txt](https://github.com/mavlink/qgroundcontrol/files/483881/qgclog.txt)

DonLakeFlyer, 2016-09-21:
Ok. Can you do the following:
0) Start with no telemetry radio plugged in
1) Boot QGC. Settings/General - Check "Clear all settings on next start" under Miscelanneous. Shut down QGC.
2) Boot QGC. Settings/Console - Turn on LinkManagerLog, LinkManagerVerboseLog, SerialLinkLog. Stay on this page.
3) Plug in your telemetry radio. You should be prompted to connect it to QGC by Android (unless you picked Always somewhere along the way). If you aren't prompted you'll need to clear the setting from the QGC app in Android settings.
4) Send the log output again

bestpest, 2016-09-22:
[qgclog.txt](https://github.com/mavlink/qgroundcontrol/files/486302/qgclog.txt)
Did as you say nothing changes. When i plug in telemetry got prompt to what i should use baby default. Choose qgc. Don't have tower and 3dr services.
Looks like qgc doesn't see the connected telemetry.

DonLakeFlyer, 2016-09-22:
Ok, so the radio never shows up as being seen by QGC. 

@dogmaphobic We have a pretty broad catch-all for radios don't we? The radio never shows up in the log. I've verified with my own tablet that with a 3DR radio + LinkManagerVerboseLog you can see full details of the radio being connected.

bestpest, 2016-09-23:
Just tested this radio on Nexus 9 lte with latest OS and old cheap Chinese tablet with 4.4 and qgc works.
I can see USB port . Is it 5.0 problem or lg g2 itself? Very odd.

DonLakeFlyer, 2016-09-23:
@dogmaphobic You have a Nexus 9 don't you?

dogmaphobic, 2016-09-23:
Yep, all "Nexus" in fact (4, 5, 7 and 9 -- all off eBay just for this)

dogmaphobic, 2016-09-23:
I plugged the radio to the Nexus 9 and it immediately asked what to launch. I selected QGC, it launched and it found the radio just fine. It's currently running Android 5.1.1 though I've run it in the past with anything from 4.x  to 6.x:

[image]

dagar, 2016-09-23:
I can also confirm a 3DR radio on a Nexus 9 (Android 7.0) just works.

DonLakeFlyer, 2016-09-23:
@bestpest Just to be double-sure. You are being prompted to allow QGC to use the radio by Android right?

DonLakeFlyer, 2016-09-23:
@dogmaphobic Does android base allowed connections on vendor id? Does it have a universal FTDI fallback like desktop does? Wondering if we need to add new ids?

dogmaphobic, 2016-09-23:
Android is a little different. From what I remember, it's up to the "driver" (Java spaghetti). That's who decides, based on the Manufacturer and Device ID what is or isn't a supported device. That's where you would add the ID. With that said, I haven't looked into that code in quite some time.

Here are the current supported devices: https://github.com/mavlink/qgroundcontrol/blob/master/android/src/com/hoho/android/usbserial/driver/UsbId.java

This is what I get with the unbranded Chinese radio I'm experimenting with:

[image]

dogmaphobic, 2016-09-23:
While looking at the log, it would't hurt to strip off the path down to "qgroundcontrol/src/comm/..." A lot of unnecessary repeated strings. No idea how to do that though...

bestpest, 2016-09-24:
@DonLakeFlyer yes. When I plug radio in, Android asks should I use qgc by default for this device.
But after that qgc doesn't see any ports. Third attached pic shows this. No device detected.

DonLakeFlyer, 2016-09-24:
@bestpest Ok, so you are saying it works for you on one of your tablets running 4.4, but it doesn't work for you on you tablet running? 

Gus, that would seem to indicator a problem in the serial port code as opposed to device id detection logic wouldn't it?

bestpest, 2016-09-25:
@DonLakeFlyer yep, it runs on Nexus 9 lte with 7.0, runs on Chinese tablet with 4.4, but doesn't run on my phone lg g2 with 5.0.1. But tower runs on the phone no issues. So in summary I have to say that it is something with qgc.

DonLakeFlyer, 2016-10-10:
Do you guys have the ability to install an APK onto your android devices. That is how I'd give you a private build to test with.

bestpest, 2016-10-11:
@donlakeflyer yes. I have.

DonLakeFlyer, 2016-10-11:
Ok. I'll get you an apk tomorrow that will tell us whether QGC code is seeing bit still coming in on the link after the disconnect.

DonLakeFlyer, 2016-10-11:
Ok, install this test build: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/OneOffBuilds/QtApp-debug.apk.

It has a big red or green rectangle in the toolbar. If it goes red while the vehicle is connected it means it is getting no bytes over the serial link. The question I"m wondering about is whether when the problem happens all bytes stop coming over the link. Or if maybe somehow the bytes are still coming by corrupted/incorrect.

DonLakeFlyer, 2016-10-11:
Sorry the above comments are in the wrong place. No need to install apk.

DonLakeFlyer, 2016-10-11:
Ok, here is the right thing. Can you do this:
* Install: https://play.google.com/store/apps/details?id=aws.apps.usbDeviceEnumerator
* Plug in your radio
* Open the usb info app and take a screenshot of the radio device info a post here

bestpest, 2016-10-12:
[image]

bestpest, 2016-10-12:
[image]
[image]
[image]
[image]

bestpest, 2016-10-12:
That's why i got when connect usb radio

DonLakeFlyer, 2016-10-12:
@dogmaphobic I believe this is because the serial ids are not in the list which the android code knows. Since android will prompt when you plug in a device to connect to QGC should we just let all USB devices connect through the QGC code?

DonLakeFlyer, 2016-10-12:
@bestpest Just letting you know where we are at with this. I know what the problem is. Working my way through some other android link testing at the moment. If that goes well I'll include the fix for this in that pull.

bestpest, 2016-10-13:
@DonLakeFlyer that's good to know. I'm watching and waiting to test it.

DonLakeFlyer, 2016-10-13:
@bestpest Can you try installing this apk: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/OneOffBuilds/QtApp-debug.apk.

Change is not fully baked yet. But should be good enough to test connection. Should auto-connect when you plug in the radio.

bestpest, 2016-10-14:
[image]
Looks like it connects but can't get any information from the copter. And comm links settings still no serial port in drop down menu.

DonLakeFlyer, 2016-10-14:
I think since the port auto-connects it won't show up in the dropdown. The parameters not loading is a separate problem not related to connection. You're running ArduPilot right? It seems to hit that more often. If you don't get a "Communication lost" and things like the HUD continue to work than communication is fine.

What order did you do things? Boot QGC, then plug in or start with no QGC running then plug in which then in turn boots QGC. Whichever could you try the reverse?

bestpest, 2016-10-14:
@DonLakeFlyer  tried both way. Same thing. Can't load parameters. Ardupilot with arducopter

DonLakeFlyer, 2016-10-14:
What version of ArduCopter? I'll give it a try.

bestpest, 2016-10-14:
@DonLakeFlyer  frame type quadrotor x, firmware 3.2.1

DonLakeFlyer, 2016-10-14:
Ok, my changes do fix the connection problem. The parameter problem you are seeing is caused by the switch to mavlink 2.0 on master. Working on that.

DonLakeFlyer, 2016-10-15:
FYI: I have all the fixes for this together now. May take a couple take to work them through to build systems.

bestpest, 2016-10-15:
@DonLakeFlyer great! Looking forward!

DonLakeFlyer, 2016-10-16:
Can you try installing again from the link. Hopefully this one will do the trick!

bestpest, 2016-10-17:
@DonLakeFlyer  no connection at all.

DonLakeFlyer, 2016-10-17:
@#()*$*(&%! Crap. Not sure why it went backwards. Let me figure out how to get some logging back from the Android side so I can figure out why it won't connect.

DonLakeFlyer, 2016-10-18:
Ok, try a new install from the apk link:
* Boot QGC
* Go to Settings/Console
* Plug in radio

On the console you should hopefully see an entry like this:
Found /dev/bus/usb/001/002:FTDI:24597:1027

Let me know what the entry is.

bestpest, 2016-10-19:
[image]
Yep. Now it seems no problem to connect and get information from copter. Thanks!
When will this be released on the store?

DonLakeFlyer, 2016-10-19:
I'll try to clean it up tomorrow and push it through.

Can you put an separate issue on the parameter missing problem. QGC should be able to run with Copter 3.2.

bestpest, 2016-10-19:
It runs but always says that warning. I will try to put new issue.

closed 2016-10-20 by DonLakeFlyer

keiler2018, 2020-10-27:
Android builds only support a single serial connection.     

How to solve this bug？

DonLakeFlyer, 2020-10-27:
> Android builds only support a single serial connection.
> 
> How to solve this bug？

You don't. You can use non-mobile device build which supports multiple connections.

PR #4157 (2016-10-20): Use FTDI driver provided by ftdichip.com
- Fixes lost connections with FTDI radios
- Modified code to only test vendor id, all product ids are let through
- Modified code to pass log output through to C++ side QGCLoggingCategory system

Fixes #4055, #3889