mavlink/qgroundcontrol #4056: Android Google Play beta broken

dagar, 2016-09-18:
I updated the Android versionCode scheme to make small point releases like v3.0.1 possible and broke beta in the process. The beta track will only accept new uploads with a versionCode greater than production. 

I thought this may be an opportune time to reconsider what we actually want.
Do we want users directly running master getting multiple updates per day straight to their phones? A lot of people are still on the beta track (potentially unaware). Several users have had issues/complaints that were resolved by instructing them to switch to stable.
Would it make more sense to use the beta track for the brief period where we stabilize and test the next release? Like v3.0.1
We could then either have a completely separate app listing for `QGroundControl Beta` or direct them to the master download to sideload the app. I believe stopping beta in the current listing will drop those users to stable/production.

dogmaphobic, 2016-09-18:
For what's worth, iOS App Store is not allowing uploading 3.0.0 as this is the same *current* approved build (even thought the build number is different). The version will have to be bumped up.

dagar, 2016-09-18:
Are you going to keep going with TestFlight once the app is in the store?

dogmaphobic, 2016-09-18:
Not sure about that. It would be up to @DonLakeFlyer. The issue above however, is independent of TestFlight.

DonLakeFlyer, 2016-09-19:
> Are you going to keep going with TestFlight once the app is in the store?

Yup.

DonLakeFlyer, 2016-10-20:
@dagar Fixed now?

dagar, 2016-10-20:
Yes it's fixed, but the way the versioning works you have to do this nop merge from v3.0.2 to master so that the next daily build automatically gets a larger version number.
https://github.com/mavlink/qgroundcontrol/commit/3950327eb671c7b6f8ffd1cc9c5a595a9a9a6327

Someday we can have dual play store listings as previously discussed.

closed 2016-10-20 by dagar